rodizio1/EZ-WifiBroadcast #1: OSD cell count has to be set manually

rodizio1, 2017-01-04:
from user geofrancis on Rcgroups:

> on the OSD i noticed that the cell count has to be set manually, this doesnt really work when you have aircraft that use 3,4 and 6 cells. 
> 
> would it be possible to have it just use the initial start up voltage to set the cell count. 
> 6-8.9v for 2s
> 9-13v for 3s
> 13.1-16.9v for 4s
> 17-19.8 v for 5s 
> 19.9-25.4v for 6s

rodizio1, 2017-04-21:
feature did not make it into 1.5, due for next release ...

geofrancis, 2018-05-23:
did you get around to looking at this?

bortek, 2018-11-02:
I am trying to clean up the clutter @careyer , do you know if this is still relevant or can we close it ?

rodizio1, 2018-11-07:
There is one thing that needs to be re-designed first: 

The OSD gets killed and re-started under certain circumstances (for example, when the wbc status message tool is being run to display  "Ethernet connected" and other messages). 

This would interfere which such functionality and lead to wrong cellcount when this OSD re-start happens after the cells have been discharged a bit already.