rodizio1/EZ-WifiBroadcast #4: Raspberry Pi Internal NIC Channel 13 not working

JulianBerger, 2017-01-14:
The channel 13 of the Internal Raspberry Pi 3 Wifi Card is not usable.

bortek, 2017-01-15:
I don't know how this is relevant to this project. Please correct me if I am wrong but ASAIK we are not using built-in Wifi adapter of PI3. Wbc uses external Wifi adapters instead of built in one.

JulianBerger, 2017-01-15:
You can use the internal module for the wifi hotspot...

bortek, 2017-01-16:
I see your point. external card for WBC and internal one for the Wifi hotspot. And there seems like you cannot use CH13 of the wifi bands. Not sure if EZ-WifiBroadcast code does anything with internal card. If not then it is Pi/Raspbian issue.  I let @rodizio1 comment on this.

rodizio1, 2017-01-16:
Have tested it now. It's indeed a bug. 

The internal Raspberry wifi card firmware does not allow channel 12 and 13 although the kernel CRDA config has all restrictions removed. It seems though, that explicitly setting "DE" as country makes the firmware allow these channels. Still need to test if that has any impact on channel or transmit power settings with Ralink and Atheros cards used for wifibroadcast transmission though. But I doubt it.

If you want to fix it yourself until the next EZ-Wifibroadcast version is released, you can edit "/root/.profile", find the "detect_nics" function and add "iw reg set DE" there.

So that it looks like this:
```
function detect_nics {
	tmessage "Setting up wifi cards ... "
	echo

	iw reg set DE

	NUM_CARDS=-1
	NICSWL=`ls /sys/class/net | nice grep wlan`

	for NIC in $NICSWL
	do
	    # re-name wifi interface to MAC address
	    NAME=`cat /sys/class/net/$NIC/address`
	    ip link set $NIC name ${NAME//:}
	    let "NUM_CARDS++"
	    sleep 0.1
	done
```

rodizio1, 2017-04-21:
fixed in version 1.5

closed 2017-04-21 by rodizio1