rodizio1/EZ-WifiBroadcast #5: Allow telemetry from UDP sources on TX

yconst, 2017-01-20:
Users of Linux autopilots (such as the NAVIO) may wish to have telemetry coming from UDP forwarded to RX (e.g. from a local (TX) instance of Ardupilot). Is this possible currently? If not, how easy is it to implement it?

Thanks

rodizio1, 2017-01-21:
It shouldn't be that hard to get the telemetry forwarded, could probably achieved by putting socat between the Ardupilot instance and and the wbc tx.

The bigger problem would be running Ardupilot and Wifibroadcast together. In general, I think running a FC which requires strict timing on Linux on a Pi is not going to be easy. Especially when there is other stuff being run already like raspivid and the wbc tx etc. I'd rather use a FC with a dedicated STM32 processor for running Ardupilot.

EZ-Wifibroadcast doesn't have a realtime kernel (and possibly other patches the Emlid guys made?), so I'm not sure how good Ardupilot will run on an EZ-Wifibroadcast image. You could either take the Emlid realtime kernel and patch the EZ-Wifibroadcast stuff into it, or take the EZWBC kernel and patch the realtime and Emlid stuff into it. Not sure what's the best approach.

yconst, 2017-01-23:
Thanks for the reply. Indeed, now that I think of it, it makes more sense to have this functionality into the Linux board-specific image (NAVIO in this case). Up to you if you want to keep this open or close it.

btw, Ardupilot and Wifibroadcast run fine in the latest EMLID beta image (they've included WBC) on a RPi2/NAVIO2.

rodizio1, 2017-01-30:
Don't have any emlid or APM hardware, so I can't do much.

closed 2017-01-30 by rodizio1