rodizio1/EZ-WifiBroadcast #6: Build guide/script

Grawp, 2017-01-25:
Hi. I've searched the wiki for a guide or a script to recreate the release image from scratch but haven't found any.
Would it be possible to add it?

bortek, 2017-01-25:
Right now it is only @rodizio1 who is building the release image as well as most of the code. Indeed it would be nice to have the codebase that "is changing and is being developed" up here on github and then some method of adding it to the base image(s). There are many peaces of the that are in different places on the filesystem. 

Could it be possible to do it like this? -> At some point in time (e.g. next release) we take an image and call it base image and all the changes from that point add to local git repo on the filesystem and then push upp here to github.

If you have other ideas or method known feel free to share.

Grawp, 2017-01-25:
@bortek I don't see how it solves parts which must be compiled (maybe I've just misunderstood it), but the basic idea is valid. My modification of your proposal:

Add a link to the specific base image to the README. It is versioned with other files so there's no problem changing it from time to time. The image should be official raspbian image or something reproducible without EZ-WifiBroadcast related changes.

Add folder with configuration files, python scripts etc... Here come several possibilities to mind:
* It might contain whole folder structure like overlay/etc/, overlay/var/lib/ and install script would just copy overlay/* to /
* It might contain just separate files and install script would copy the files to the correct places.
* It might be a bunch of .patch files and install script would contain lines like `patch / < logind.patch`

And finally install script can run make on the compilable parts of the project and put the compiled files to their proper places.

mzemb, 2017-01-26:
@Grawp It sounds like a good way to go. I prefer your options 1 or 2, as 3 will be less readable in my humble opinion.
I suggest to make use of a loop mount to modify the base image.

rodizio1, 2017-01-30:
Sorry, don't really see the point in putting a lot of extra time into creating and testing image builder or overlay scripts. I rather use that time for improving it and adding features. In the long run buildroot would be better anyway.

I'll add all the sources included in the image to Github though, just give me some time ...

mzemb, 2017-01-30:
Indeed, buildroot seems a nice way to go.

Grawp, 2017-01-30:
@rodizio1 I've also wanted the script, guide or just list of changes, because I want to try to package the WifiBroadcast as ArchLinuxARM packages (one for kernel, one for the app itself, one for network and other settings...).

Grawp, 2017-01-31:
@rodizio1 I've also wanted the script, guide or just list of changes, because I want to build my own WifiBroadcast setup.
(Try to package the WifiBroadcast as ArchLinuxARM packages: one for kernel, one for the app itself, one for network and other settings....)

MikeStros, 2017-04-05:
@rodizio1 
Hello!
Please, tell me what distributive is used as base for EZ-WiFiBroadcast images? I really like starting speed and nice splash instead of console output. I want to do similar thing in my smart home project. 
I think I can use EZ-WiFiBroadcast image disabling rx/tx scripts and adding my custom apps, but I don't know how to change splash.
Thank you!

bortek, 2017-04-05:
It is Raspbian operating system (what you cal "distributive") from Raspberry Pi foundation. :) Hope that helps.

closed 2017-04-05 by bortek