rodizio1/EZ-WifiBroadcast #8: Build guide for deploying on targets other than raspi

hatsunearu, 2017-03-08:
I would like to use one of my spare x86 laptops as a ground station instead of a raspi so that I don't have to lug around a monitor or suffer additional latency with forwarding. I'm wondering if there are any guides to deploy on something like this.

bortek, 2017-03-08:
This project is solely for Raspberry Pi and I am not aware of any port for x86 or other architectures. We have no plans for that either, I am sorry.

closed 2017-03-08 by bortek

johnboiles, 2018-11-30:
I think the Linux kernel patches would probably work for x86 Linux as well. This would probably be a matter of starting with a version of Debian on a comparable Linux kernel to what this project has. Compiling the Linux kernel with the patches from this project, removing the references to wiringpi. And finally recompiling the software for x86. @bortek are there any reasons you can think of where this wouldn’t work? Is there any other pi specific stuff that’d need to be worked around?

RespawnDespair, 2018-11-30:
Regular Linux as a GROUNDSTATION has already been done several times. 
The transmitter is relying on the raspivid to use the raspberry pi camera, so the only issue is the AIR side.
In the forum thread user1321 posted a working x86 Linux groundstation although i'm not sure what he did exactly. Running the kernel patches would allow you to build an optimized x86 kernel for sure. The most important parts are the driver patches.

rodizio1, 2018-11-30:
The kernel patches as well as the wifibroadcast stuff will also work on a PC, yes. Like RespawnDespair wrote, PC as groundstation has been done, but it's not the same as on the Pi. You need to use something else than hello_video.bin (which only works with the Pi GPU) for video display (most people seem to use gstreamer) and also I'm not sure if the OSD can run on a PC (it also uses the GPU).

>I would like to use one of my spare x86 laptops as a ground station instead of a raspi so that I don't have to lug around a monitor or suffer additional latency with forwarding. I'm wondering if there are any guides to deploy on something like this.

Actually, the forwarding doesn't add much latency, maybe 5ms, if it all. For lowest latency, use the FPV_VR app with a fast Google Daydream Ready smartphone, it'll give _less_ latency than most HDMI monitors connected to the Pi.

hatsunearu, 2018-12-01:
It's been a very long time since I tried this, but I vaguely remember using gstreamer on my ubuntu laptop to get the stream working.

johnboiles, 2018-12-01:
I wonder if this would work in a Linux VM on Mac or Windows. Seems like it should. But I'm not sure if there's some property of the USB VM passthrough that could throw a wrench in things.

johnboiles, 2018-12-02:
@RespawnDespair thanks for pointing to @user1321. I think you're talking about [this thread](https://github.com/rodizio1/EZ-WifiBroadcast/issues/108) which was extremely helpful. Thanks for the tip.

I'm having some success running this in a VM, just working on piping out the video now. Maybe I can put together a Vagrantfile or VDI image of it for distribution to others. Two questions

1. What functionality do the kernel patches add? Looks like they increase the tx power, and maybe enable some extra channels. Anything else I'm missing?
2. Has anyone seen a solution for rendering the OSD when running on the laptop? I suppose I could write a simple program to show the values. I'm mostly interested in signal strength