rodizio1/EZ-WifiBroadcast #10: Please clarify license terms of the EZ-WifiBroadcast images and provide full source.

hsteinhaus, 2017-04-24:
The offered binary images seem to be based on lots of GPL v2 and GPL v3 software components. Some  of them (e.g. the Linux kernel) are obviously subject to source code changes by the EZ-WifiBroadcast image build process. However, this git repository does not contain all code/script/tools necessary for re-building the EZ-Wifibroadcast images by third parties. I am neither a lawyer nor a (substantial) holder of copyrights in those underlying components, but I think offering binary derived work of GPL software without fulfilling the license requirements is probably prone to legal trouble sooner or later.

Please fix these problems by clarifying the license terms of the EZ-WifiBroadcast images provided and by fulfilling the license requirements of all sub-components you are deriving from. Offering the bare patches is IMHO not adequate to ensure this.

bortek, 2017-04-24:
Thanks for pointing it out. I will will need to read up on the license terms and think how to resolve this in a proper way.

rodizio1, 2017-04-24:
> The offered binary images seem to be based on lots of GPL v2 and GPL v3 software components.
Some of them (e.g. the Linux kernel) are obviously subject to source code changes

The source code changes to the kernel are available in the patches directory (Granted, not the ones from v1.5, still haven't found time to create them ...).

> by the EZ-WifiBroadcast image build process.

The "image build process" is me booting-up the image on a Raspberry, making changes to the system, then using dd to pull an image. There is no build script or other code that I could share.

> However, this git repository does not contain all code/script/tools necessary for re-building the EZ-Wifibroadcast images by third parties.

I'm not obliged in any way to make it easy for third parties to re-build the image. Also, I'm not obliged to include any tools to do so. Like I wrote above, I don't even re-build the image myself. If you want to "re-build" it, download it, make desired changes and pull an image of it. If you want to make your own image, take the sources from the EZ-Wifibroadcast image and use them for your image.

The image consists of the following parts:

1. The Raspbian Distribution: Source code is publicly available, changes I made to the system can be looked up in the image. Original config files are included with file extension .ORIG.
2. The Linux Kernel (part of the Raspbian Distribution) and my changes to it: Source code is publicly available, my changes to the sources are available in the patches directory.
3. The Atheros firmware: Source code is publicly available, patches to the source I made are exactly the same as Befinitiv. Those patches are also publicly available on Befinitiv's Bitbucket page.
4. Various extra tools not included in the Raspbian distribution: Sources are available publicly. In cases where changes to the sources have been made, full sources are included in the image.
5. Befinitiv's wifibroadast and my changes to it: Full sources are included in the image.
6. Rangarid's Wifibroadcast_osd and my changes to it: Full sources are included in the image.
7. mmormotas hello_video.bin: Original source code publicly available, full source included in the image.

8. Software I wrote:
a) Tools based on existing software: Original sources are publicly available, full sources of my software are included in the image. Original software and it's authors are mentioned in the sources.
b) Software I wrote myself, "from scratch": Sources are included in the image.

> Please fix these problems by clarifying the license terms of the EZ-WifiBroadcast images provided and by fulfilling the license requirements of all sub-components you are deriving from. Offering the bare patches is IMHO not adequate to ensure this.

I think they are all fullfilled. Also, I don't think that I need to "clarify" anything. You can look up the licensing in the image, I did not delete or change any of the original license or copyright notes.

BTW: As far as I know, I'm not even required to offer the sources in the image or on Github.  They just need to be made available. That can also mean sending you a CDROM with the sources for a small fee ...

bortek, 2017-04-24:
I read up on GPL license and think that this point is over exaggerated. This is not even a commercial project, its a hobby project for enthusiasts. So don't get too excited :)

GPL license in general boils down to publishing the source code which as rodizio mentioned is provided in this repo, apart from v1.5 which will be added later on. Besides, the GPL license does not mention anything about ISO images which what this project delivers to the end user. GPL license mentions "binary" or rather "program binary" and if ISO image is counted as a binary program is a subjective matter.

To satisfy license lovers what we can do is to create a short document like "terms of usage of the EZ-WifiBroadcast" consisting of the numbered bullet points that rodizio  mentioned (with some more links to the included parts) . That would more than suffice.

rodizio1, 2017-04-25:
Yeah, a wiki page or something that lists the software used for this project would make sense. I'll do that.

bortek, 2017-05-16:
A new page has been created on Wiki called "Software used in the image" which summarizes software used in the image.

closed 2017-05-16 by bortek

az0uz, 2017-05-16:
Could you also add an open source licence to the project ? Would be nice to be allowed to use this code openly, I think that without a LICENCE file at the root of the repo, all the new scripts en codes are not openly usable. Thanks a lot

bortek, 2017-05-16:
As mentioned above the code consists of many parts from various codes and sources like for example Raspbian thus the project cannot be license stamped in that sense. In that case only the "added" code can be license stamped but it wont make much difference.

az0uz, 2017-05-16:
Yes I am talking about adding a licence for the modifications to wifi broadcast and the automatic configure script (mainly /root/.profile)

rodizio1, 2017-06-17:
/root/.profile is part of Raspbian, so as I understand it, it has the same license as the Rasbpian distribution.