rodizio1/EZ-WifiBroadcast #11: Build does not work for wifibroadcast

Hyphen90, 2017-05-08:
Hi, there are duplicated symbols in the build - for example:

rx_lib.o:(.data+0x0): multiple definition of `max_block_num'
rx.o:(.data+0x10): first defined here

Please can you clarify/fix it?
And are all the changes for the 1.5 version included in this repository?

rodizio1, 2017-05-12:
Sorry, the code in the Github Repo is outdated, I have only maintained the wiki lately. The source in the image is the latest version. I'll see that I upload v1.5 code and also the kernel patches on the weekend ...

titeuf007, 2017-05-13:
thanks rodizio for this great project
i hope you will continous to improve it! you are the one project i like most with arducopter and betaflight!
the only one thing missing in this great project is Automatic frequency hopping to be complet
have a nice day

bortek, 2017-05-16:
I keep this open until you have time to add the code.

terial, 2017-05-28:
hey @rodizio1 any updates on getting the code updated? thanks!

rodizio1, 2017-06-17:
I have updated the sources on Github to the 1.5 sources now.

closed 2017-07-08 by rodizio1