rodizio1/EZ-WifiBroadcast #12: Enhancements...

Toni-42, 2017-05-09:
Hello,

It’s great OSD/Wifibroadcast-Implementation! Wow!

As for OSD i have done something like that for my own FC (based on AVR), but in BASIC, so it's hard for me to participate directly.

I have 3 enhancements that are very easy to implement.

1.    It would be nice if the home position (LTM protocol origin frame (O)) would constantly updated (the program does not do this, even if new home latitude and longitude values are sent via O-frame).
This would allow to use for intermediate gps-points and see the "Arrow" and "distance" current values in autonomus fly.

2.    It would also be good to make the orientation of the Home Arrow relatively. This means to take the heading into account.
For example, if Home Arrow points to east and Compass (heading) points to west, it makes sense to point the arrow down (180°) and not to 0°. So the Pilot just folows the arrow to the next GPS-Point/Homepos...
    
Just bring Heading (normaly 0...359°) and Course (normaly 0...359°) in the following Coordinate System:
[image]

If Heading > 180 Then
Heading_new_coord_sys = Heading - 360
End If

If Course > 180 Then //in your Implementation should be Course=Angle_home_arrow
Course_new_coord_sys = Course - 360
End If

The relative angle of Home Arrow is then:
Relative_angle_home_arrow = Course_new_coord_sys - Heading_new_coord_sys

This should be adjustable in osdconfig.txt ... for people who don't like this(?)

3.    Perhaps you can also display the units [m], [km/h] also with altitude and speed values?

Greetings
Toni

rodizio1, 2017-05-12:
These suggestions sound good (and not too much work ...) I'll look into that when I find some time.

bortek, 2017-05-16:
Adding ti to 1.6 release if you have time otherwise we move it forward later on.

rodizio1, 2018-02-12:
2. and 3. have been implemented in 1.6RC5.

1. Will be added later.

rodizio1, 2018-04-02:
Finally, all three points are implemented in 1.6RC6

closed 2018-04-02 by rodizio1