rodizio1/EZ-WifiBroadcast #13: no clear instructions

amin005, 2017-05-11:
hello I would like to setup EZ WB  in Orange pi zero  single board computer but there is no clear instruction for that  could you please write one for us non raspberry pi  users  so it could be a universal Linux project  
thankk you

pilotnbr1, 2017-05-11:
Looks like the orange pi zero does not have a camera...So it won't work as a tx. Also for a tx h265/h264 gpu encoding decoding is almost a requirement to ensure low latency- not sure if the orange pi zero checks that box.

amin005, 2017-05-11:
the  orange pi zero plus  board with Allwinner H3 Feature Ultra HD 4k and Full HD 1080p video decoding  and H.264 High Profile 1080P@30fps encoding   its price around 15 $  board is low profile also some version has csi bus and 5$ 2megabixl camera also there are some AR9271 Wireless dongles with 100mw for like 6$ the full system would be under 50$  sorry if this is not the suitable place to address that .

pilotnbr1, 2017-05-11:
Well based on that, all the hardware seems to have potential to be a good TX. The wifi chipset is intriguing as it may have the ability to be put into monitor mode (like other AR9271). Personally I would not use this board as an RX as it has too few USB ports- you get a much better image running diversity with multiple wifi adapters. 

The big question is driver support... You would have some work getting this going. I would actually start by googling/reading befinitiv wifibroadcast blog. Start with one of his images and modify from there.

The main forum is here
https://www.rcgroups.com/forums/showthread.php?2664393-EZ-WifiBroadcast-cheap-digital-HD-transmission-made-easy%21

rodizio1, 2017-05-12:
Like Pilotnbr1 already wrote, it's doable but would be quite some work, a lot of things are different on the Orange boards. I have no plans on supporting other boards in the near future.

closed 2017-05-16 by bortek