rodizio1/EZ-WifiBroadcast #14: Console access

josemidevel, 2017-05-11:
Very nice work!!!

Just a questíon. My pi 3 could perform more task. Could be possible access to te console for script installation? (For example for arduino communication to perform some tasks)

Thanks in avance.

rodizio1, 2017-05-12:
Yes, you can for example connect the Raspberrys to you home router via Ethernet (it will show the IP it has acquired on the HDMI display). Or also via Wifi hotspot.

Login via SSH with username "root", password "raspberry", type "rw" to make filesystem writeable and make your changes. "Main script" is /root/.profile.

Be careful though, if you load it too much with other tasks, (even though CPU load may not be high) you might get badblocks or other problems.

josemidevel, 2017-05-12:
Thanks for the info and your quick response.

closed 2017-05-16 by bortek