rodizio1/EZ-WifiBroadcast #15: Pi Zero W and Camera V2.1

Hyphen90, 2017-05-13:
Hi,

when I burn the 1.5 image to a SD card, everything will start up, but the camera is not detected (supported: 1, detected: 0). I read the hint, that I had to copy the files from the directory "pi0w-firmware" to the root of the card, but then nothing happens (on HDMI or LED). Why I have to do it, when the system starts up with the original files?
And must I power the  AWUS036NHA from an external power source (breakout-cable), or is the current from the micro USB connector enough under full TX load? It is recognized and seems to be working.

Thanks!

rodizio1, 2017-05-14:
It turned out that the latest firmware in the pi0w-firmware directory does not work with the kernel in the image. The 2017-03-03 firmware version works though.

I have updated the instructions on how to add Pi0 W support here now: https://github.com/bortek/EZ-WifiBroadcast/wiki/Pi-Zero-W-support

As for the cable, I'd strongly recommend soldering it directly to the power supply (and keep it short) like in the photos shown under wiring instructions in the wiki. Even if it seems to work on first sight, you may get all kinds of issues.

titeuf007, 2017-05-14:
rodizio thanks for you support!!

Hyphen90, 2017-05-15:
Yes, thank you very much for the updated instructions. I'm totally new to Raspian and a little bit helpless, instead looking forward to my first flight :-)
That the USB port could not drive at least the minimum 500mA (2,5W) for the peaks, even with an additional low ESR capacitor, is not so nice. And also thanks for this investigation, that you save others from such troubles.

closed 2017-05-16 by bortek