rodizio1/EZ-WifiBroadcast #16: New device/drivers compatibility

az0uz, 2017-05-15:
Hi there,
I got wifibroadcast working with the Alfa AWUS036AC (chipset RTL8812).
For now, it needs a custom compiles kernel module and I only got it work well in 5GHz. In 2.4Ghz my tx throughtput is capped to ~800kbps and I did not managed to have a good quality/fluidity/latency (is that the expected baudrate for 2.4GHz ? why don't I have more like 54Mbps with wifi n ?)
The wifi card can also transmit at 40dbm instead of 30dbm on both bands.
For the 5Ghz I tuned it to have <200ms latency, 720p 45fps.
I would like to share my method, how can I contribute ?
Could you have an image builder script like the one done by befi on the original wifibroadcast so the community can help develop this project ?

I am gonna try making compatible more common USB sticks like rtl8192 (best seller RPi wifi sticks) it could be nice to have a cheap way of adding diversity, I already got it on ubuntu for the rx, it works out of the box :) so it seems it's a matter of installing/compiling/tweaking the right drivers on the RPi

Cheers, thanks for the nice bundle  \o/

rodizio1, 2017-05-15:
That's great. Unfortunately, there is no build script for the image and I have no immediate plans to make one. But it already helps a lot to know that these cards work, thanks for the info. What drivers did you use? If it's not too much work and monitor mode turns out to be stable I'll integrate it in a future release.

> In 2.4Ghz my tx throughtput is capped to ~800kbps and I did not managed to have a good quality/fluidity/latency (is that the expected baudrate for 2.4GHz ? why don't I have more like 54Mbps with wifi n ?)

Could be that they inject slower than other cards for whatever reason, or could simply be that the channel you injected on was already in use.

Did you set the bitrate manually? Or do the Realtek drivers support setting bitrate via Radiotap headers like the Ralink cards (controlled via the -d parameter of the wbc tx)?

Testing throughput can be done with this command for example:
`cat /dev/zero | pv -i 0.2 -r | /root/wifibroadcast/tx -p 0 -b 8 -r 4 -f 1024 -t 1 -d 18 -y 0 wlan0 >> /dev/null`

With Atheros cards on 18Mbit 11g datarate I get 1.05Mbyte/s, on 24Mbit 11g about 1.27Mbyte/s (with 50Ohm dummy load on the card to simulate a clear channel).

bortek, 2017-05-16:
@az0uz IN the future please post your question in the forum. thanks.

rodizio1, 2017-05-17:
Shouldn't we handle feature requests and contributions also here? EZWBC is still not really "developer-friendly", but I'll add some more sources to Github soon (don't have much time right now, sorry).

Az0us, regarding the build script again: I just took Raspbian because it was convenient and gave fast results. I think it would make more sense to use buildroot, this would make it "buildable" automatically anyway and be much smaller and better to maintain (no hassle with systemd and all that other stuff running on a "full" distribution).

az0uz, 2017-05-18:
So after real testing, Alfa AWUS036AC works great, I haven't tested the maximum range but it's >150m for sure. though only on the 5GHz band, on the 2.4GHz one there are lot's of glitches. But this may be related to your comment about only a 5GHz amplifier being present.

For the Alfa rtl8812au driver (needed for AWUS026AC) I did the right change to compile it on raspberry pi from the one I found that was working. it here: https://github.com/az0uz/rtl8812au/
then you can:
```
sudo ip link set wlan1 down
sudo iw wlan1 set monitor otherbss fcsfail
sudo ip link set wlan1 up
sudo iw wlan1 set freq 5765 #or: sudo iw wlan1 set channel 153
sudo iw wlan1 set txpower fixed 4000
```

I got the rtl8192cu working on ubuntu for receiving, It's very useful to have diversity on the 2.4GHz band because those one are very cheap and easy to find.
The driver is in fact already compiled in the latest raspbian linux stable build (I am on kernel version v4.9.27-v7) but not loaded by default.
I read there was some problem in the past with the rtlwifi_new drivers used in most distribution today so they had to use an old version of the 8192cu driver. Though they are starting to include the upstream driver for people to try it now (it support monitor mode). So it would need extensive testing.
In nutshell you need to remove 8192cu and load rtl8192cu that's it.
```
sudo modprobe -r 8192cu
sudo modprobe -a rtl8192cu
sudo ip link set wlan1 down
sudo iw wlan1 set monitor otherbss fcsfail
sudo ip link set wlan1 up
sudo iw wlan1 set freq 2437
sudo iw wlan1 set txpower fixed 2000
```
I don't have any problem with limited bandwidth since I switch to this project version of wifibroadcast and using the `-d` option. I was originally using the befi version of wbc from He's bitbucket repo.
on the rtl8192cu, with the command `cat /dev/zero | pv -i 0.2 -r | sudo /home/pi/wifibroadcast/core/tx -p 0 -b 8 -r 4 -f 1024 -t 1 -d 18 -y 0 wlan1 >> /dev/null` I got 3.5MiB/s
It doesn't seems to be limited to 18Mbps now but I guess the -d command doesn't work with that drivers, I will need to investigate further.
It's not fully tested yet, I'll give update as soon as I have time to try it.

I didn't know buildroot, I am working on raspbian lite right now. it's nice because it removes all the UI from the raspbian distribution and is very lightweight.

I think it would be nice in fact to implement wbc with systemd, it would help open source development from the community by adding modularity clarity and standardisation to the code. And systemd 
implement some nice features as starting the service at the right moment (just after the net interface are loaded), instead of using sleep time and auto restarting the daemon if it exit with a failure code.

It would be nice to have something easily installable on a standard raspbian os, with a config file to create N /dev/wbc[NUMBER] as input/output nodes and have all the different. This could be very usefull for research purpose and let people easily imlpement other stuff as rtkGPS or any uplink/downlink (as suggested by https://github.com/bortek/EZ-WifiBroadcast/issues/17).

I would be glad to start developing/porting the existing code into that direction (for a v2.0 maybe ?) cause I think there is a huge potential in wifibroadcast and lot's of application there. But right now the licensing of this repo needs to be more clear before I an start spending time on it.

Hyphen90, 2017-05-19:
Using buildroot would be a perfect choice. It will take some time to dive into the details of buildroot, but when you did, it works like a charm with a fine grained small footprint and clear separation of packages and external builds.

az0uz, 2017-05-19:
Make sense to build the EZBC image with that then but what I am thinking about is more the core of wifibroadcast which should be more like a Debian package, for example I am using an ubuntu laptop that has a large anti reflection screen as a fpv video receiver. I think it would be also nice to be able to install it on any raspbian ubuntu (and maybe more) without having to run the lightway prebuilt image.
I agree that keeping a standalone "work out of the box" RPi image is very nice too. But I see several applications where it could be interesting to play with the wbc kind of communications

geofrancis, 2017-05-19:
> cause I think there is a huge potential in wifibroadcast and lot's of application there

i agree with this, especially with pixhawk machines. there has been something like this missing for a very long time, the only other thing that came close was the solo video link but that was never released. 

i can see this becoming the ultimate companion computer for a pixhawk, with hd video, osd, telemetry links rc, 2.4/5ghz so legal in every country and with a single radio taking over from the 4 or so that would otherwise be used.

az0uz, 2017-05-19:
What's an acceptable video latency for racing do you think ? <50ms ?

rodizio1, 2017-05-20:
> So after real testing, Alfa AWUS036AC works great, I haven't tested the maximum range but it's >150m for sure. though only on the 5GHz band, on the 2.4GHz one there are lot's of glitches. But this may be related to your comment about only a 5GHz amplifier being present.

Great. Glitches on 2.4Ghz are probably because of interference from other wifi networks (or did you test outside atleast 500m away from houses etc.?) You can also try to remove the antennas on the RX sticks and keep the TX right next to the RX, if glitches go away then, it's interference.

> The driver is in fact already compiled in the latest raspbian linux stable build (I am on kernel version v4.9.27-v7) but not loaded by default.
> I read there was some problem in the past with the [...]

Problem is, newer kernels than the 4.4.11 I'm using (atleast up to 4.4.32, haven't tested the 4.4.5x from the latest raspbian release or the 4.9.x versions yet) have problems with usb or multitasking or whatever, causes badblocks because packets are delayed too much. Can it be compiled for 4.4.11 somehow?

> It doesn't seems to be limited to 18Mbps now but I guess the -d command doesn't work with that drivers, I will need to investigate further.  It's not fully tested yet, I'll give update as soon as I have time to try it.

The readme of the drivers say "to inject frames with b/g rates use the Rate field in the radiotap header", so I would've thought the datarate parameter works. Hmm, maybe try setting the bitrate manually with iwconfig or iw.

> I think it would be nice in fact to implement wbc with systemd, it would help open source development from the community by adding modularity clarity and standardisation to the code. And systemd [...]

Systemd is not needed at all, it just causes issues. A lot of unnecessary functionality cannot easily be disabled, I had to delete stuff etc. because there was no proper way to disable it (journald I remember ...). That daemon auto restarting logic takes too long, the way I do it in the scripts is faster and more reliable. For example, when a wifi card on the RX is accidently removed, the script detects that and re-starts the RX with the remaining wifi cards as quickly as possible. 

Starting the service "at the right moment" is also not really possible with systemd, I have actually intentially removed befis systemd startup scripts and other stuff (including the raspbian networking setup mechanisms) because the Atheros cards crash (especially when using more than one or two) if they are not initialized the way I do it in the .profile script.

That is:
- While nothing else is accessing the USB bus at the same time
- About 1-2s after the drivers have been loaded (if you do it too soon, they crash)
- Leaving some time (0.5s or so) between ifconfig/iw commands
- Not putting them into promiscous mode

> It would be nice to have something easily installable on a standard raspbian os, with a config file to create N /dev/wbc[NUMBER] as input/output nodes and have all the different. 

The reason why it's possible to transmit 500.000 packets with less than 10 packets lost while walking through a forest with EZ-Wifibroadcast is because everything is carefully optimized and has been extensively tested exactly the way it is. If you take a standard Raspbian, you get all kinds of (not directly obvious) issues. Atheros cards won't initialize sometimes on bootup, more than two cards won't work at all, diversity doesn't work properly under certain circumstances, you get lost/delayed packets, badblocks and image stuttering as soon as some other process like systemd's journald or logrotate or whatever unncessary stuff running on standard distributions decides to use ressources. In general, the multi-tasking and the forwarding of the packets from the card over the usb bus through the kernel to userspace is not very "realtime" in linux and on the Pi, everything is timing-critical.

The same thing will likely happen if you start changing the code or how the video stream is being distributed (i.e. after the packets have been re-assembled). It uses a combination of ftee, cat and fifos/pipes. Doesn't look really clean and is definitely not the best way to do it. But it works. You can connect/disconnect stuff on the wifi-hotspot while the video keeps running. Also, throughput issues on the wifi-hotspot link will not affect the wifibroadcast link. Try using a standard tee command or "improving" it by changing the the rx.c command to write to to more than one fifo directly instead of stdout, doesn't work reliably (I've tried, look into videorx.c in the 1.4 image ...) gives badblocks. Of course, could be my lacking C-skills, but still, it would need a lot of testing, a thorough understanding how pipes and input/output with C and under Linux works, blocking/non-blocking I/O, select()/poll()/read()/write() all those functions, getting the loops inside the rx.c right so that one thing does not afffect the other (or use multi-threading, but then you are at the linux task scheduler's mercy again, i.e. uncontrollable latency/jitter). Oh, and maybe even require additional logic like local buffering in the rx process.

If you have the knowledge and time to implement real improvements, well, go ahead :) I'll try to answer any questions you may have and give as much input and hints as possible. 

[... cut ...] e on it.

I'd love to see somebody take it over and make a better maintainable buildroot-based version. 

Still have a lot of ideas that I have either no time for or lack the C-knowledge to properly implement them, would be cool to see these things being done some day.

Regarding the licensing: What _exactly_ is not clear? It's all based on GPL code anyway, so it obviously everything must be GPL also. Or am I wrong there (I'm not a lawyer)? Tell me what exactly needs to be fixed and I'll fix it.

rodizio1, 2017-05-21:
Just did some testing with an Edimax micro dongle with an RTL8192CU chipset. The 4.4.11 Raspberry kernel seems to have the rtlwifi/rtl8192cu driver commented out, had to change that in the makefile. After doing that, receiving seems to be working fine.

Injection gave the same results as you had, 3.5Mbyte, it seems to use MCS 7 65Mbit datarate. Setting bitrates via iwconfig or iw did either not change anything or made it inject with only 1Mbit bitrate.

Regarding the -d bitrate parameter: Do you have the MCS bitrate patch in your kernel on the Ubuntu machine? https://github.com/bortek/EZ-WifiBroadcast/blob/master/Patches/mac80211-radiotap-bitrate_mcs_rtscts.linux-4.4.patch

Without that, the -d parameter doesn't work (atleast not on a Raspberry 4.4.x kernel)

az0uz, 2017-05-30:
No I didn't apply the patch.
So I think I am gonna do some testing on how to implement wifi broadcast on a separate repo until I have something stable and with the same performance.
Cause it doesn't make sense having something slower just for the sake of readability I guess

closed 2017-10-20 by rodizio1