rodizio1/EZ-WifiBroadcast #17: 1.6 suggestions

geofrancis, 2017-05-18:
as well as the main telemetry link that the osd works from having some more transparent uarts would be very useful for people with gimbals, RTK gps, multiple telemetry links like frsky and mavlink or other devices that use serial ports, currently the only solution out there is to have multiple telemetry radios. wifibroadcast video uses so much bandwidth that 3x 115200 serial links should be barely noticeable 

theoretically it could also be used for transparent RC links. then you wouldn't have to worry about generating the rc output, you should just need the correct baud rate for the protocol so what comes in goes out. for failsafe you just stop the output simulating rc disconnect, most flight controllers will just see that as out of range and failsafe without having to configure anything.

sbus is 100,000
dsm and sumd is 115200

just have something in the config like:

TX
telemetry/osd port: /dev/ttyUSB0
baud1: 57600 
serial2: /dev/ttyAMA0
baud2: 100000 (sbus)
serial 3 /dev/ttyUSB1
baud3 115200

Rx 

telemetry/osd port: /dev/ttyUSB1
baud1: 57600 
serial2: /dev/ttyAMA0
baud2: 100000 (sbus)
serial 3 /dev/ttyUSB1
baud3 115200

then each serial connection can have its inputs and outputs setup, you already have something similar for  telemetry and Rc

geofrancis, 2017-06-12:
i found this and wondered if it would make it easier to impliment multiple serial ports

http://www.usriot.com/p/modbus-tcp-ethernet-ip-modules/

Montilien, 2017-07-23:
Hi, congratulations for this good work. Did you think to add 3G/4G dongle alternantive for very long range Video, telemetry, and RC???
Best regards
Andre

closed 2017-10-21 by rodizio1

Montilien, 2017-10-23:
Hi Rodizio,
Why have you closed my question without any response?????
I think that my question is interesting for many user of EZ WIFIBROADCAST...
Best regards
Andre