rodizio1/EZ-WifiBroadcast #19: Pi zero w

pilotnbr1, 2017-06-08:
The file start_x.elf is the only one that needs to be rolled back to 2017-3-3

bortek, 2017-06-12:
Mind to elaborate what this is about ? ;)

rodizio1, 2017-06-12:
I know :)

This is kinda strange. From reports on rcgroups, the Raspbian 2017-03-03 firmware should work.

The link to the firmware given in the wiki (http://downloads.raspberrypi.org/raspbian/archive/2017-03-03-17:49/boot.tar.xz) should contain exactly that firmware.

So, did the firmware files from the link above not work? You wrote in the forum that it worked after you downloaded the complete 2017-03-03 image and took the firmware from there. But that should be the same as the one in he link above (?)

Or do you mean it works only if you replace just the start_x.elf file, i.e. as soon as you also replace the others, it doesn't work anymore?

pilotnbr1, 2017-06-12:
Initially I followed the github instructions regarding zero w support and the w would briefly blink for about 10 seconds and then just go to a dim green led. I tried several times and nothing...

Per rcgroups if I started with a fresh ez 1.5 image and just swapped out the start_x.elf from the older firmware than the w boots and runs WBC normally.

pilotnbr1, 2017-06-13:
IMHO the problem is only with the instructions on the GitHub. Do not copy and replace dat files etc on the 1.5 image. It won't work. Just replace the start_x.elf file

Sorry for the cryptic one line issue- think I was drinking that night lol

rodizio1, 2017-06-17:
thanks, I'll change the Wiki accordingly.

closed 2017-07-08 by rodizio1