rodizio1/EZ-WifiBroadcast #20: AWUS052NH V2

terial, 2017-06-15:
Hi, I am trying to get two Rpi3 to connect with two AWUS052NH V2 wifi adapters. I know a little bit about linux and raspberry pis but I can not for the life of me figure out why they wot connect. I have a screen on the receiving side. My guess is that they are both set as receivers... any ideas or help would be appreciated. thanks

az0uz, 2017-06-15:
You should have error logs and screenshot saved to both SD cards (main FAT partition so readable even from windows). try posting those here so we can help you debug

terial, 2017-06-15:
Hey @az0uz thanks for the response. I checked both cards and there were no screenshots or logs. To check i started up the receiving Rpi3 without the AWUS052NH wifi dongle attached and then checked the SD card and did see a screenshot and and a text file. Once i cleared the error log and screen shot and restarted it with everything connected there were no more logs created. I have the LCD screen on the receiving Rpi3 but I am unsure how to check the transmitting Rpi3 to see if that is working as should.

terial, 2017-06-17:
i connected another screen to the TX Rpi and the last line oin startup; says Starting RX ... (FEC: 8/4/1400) How do I make it start as TX and not RX?

az0uz, 2017-06-17:
It supposed to switch to tx when it detects the raspicam, check your camera cable and the camera it self if the cable is well inserted, your camera and/or cable may be broken if it doesn't start as tx

terial, 2017-06-17:
Ok, that makes sense... I am using a logitech C920 camera instead of raspicam. I am trying to figure out how to change some things to enable the webcam but the read only file system is making things hard.

az0uz, 2017-06-17:
Yes it's not really straightforward to change to USB cam,
Use v4l2-ctl to switch the camera to the h264 720p stream with the right baudrate like 3Mbps, ne you have to modify where it uses 'vcgencmd get_camera', it uses this command to detect if the camera is plugged in or not and switch to rx/tx mode.
(Type rw to put in read write mode)

rodizio1, 2017-06-17:
Okay, so the issue is not the Alfa, but the Logitech Cam, I've renamed the issue :)

If you want to try to make that cam work, change the tx/rx detection routine like az0uz suggested (instead of running vcgencmd you may want to do something like "lsusb | grep Logitech" or whatever to check if the USB cam is connected).

Look in .profile in the tx_fuction(), there, raspivid is being run and the output piped into the wbc tx, you'd have to change that to use v4l.

terial, 2017-06-17:
to switch to v4l2-ctl is it as simple as changing all lines with raspivid to v4l2-ctl?

terial, 2017-06-17:
Thank you @az0uz and @rodizio1 this will help out! Thanks for all our hard work in helping to create another great component in the DIY community.

deksprime, 2017-06-26:
I'm also trying to achieve the exact same thing, although I have some other questions regarding the `.profile` file. Is there a way to edit the `.profile` file other than the current method which is enabling 'rw' option and changing it on the Raspberry Pi? Is it somehow possible to enable editing on the SD card and then reverting the write-read settings back to original?

az0uz, 2017-06-26:
@deksprime when you type `rw` it will remount the sdcard as read-write and when you type `ro` it will remount it back as read-only. Also as soon as you reboot the Pi, it will be back in read-only.

The best would be to switch even the raspicam to v4l2 (by just enabling the kernel module) it would allow even exposure adjust from the remote while streaming

deksprime, 2017-06-27:
@az0uz thank you for the help! I will try this and see if I can make it happen...

geofrancis, 2017-06-27:
i would be very interested in this if it would allow dual cameras, even if the logitech one has a lot of lag .

hlyx, 2017-06-27:
Hi,
very interesting good work done here.

if someone could help by summarizing the process of integrating the USB- uvc cameras (Logitech C920 and so )as im new.
thanks :)

hlyx, 2017-06-28:
Hi,
very interesting good work done here.

if someone could help by summarizing the process of integrating the USB- uvc cameras (Logitech C920 and so ) discussed here so far. and that because im new to about all fields of knowledge here.
thanks! :)

hlyx, 2017-07-09:
Does anyone has this issue?

On the Tx scree everything starts without an error, but after starting transmission:

"Warning lost connection to stdin. Please make sure that a data source is connected. "

This error is repeated endlessly. 

Thanks

closed 2017-10-20 by rodizio1