rodizio1/EZ-WifiBroadcast #21: Beta development kernel

rodizio1, 2017-06-18:
For people who would like to try the latest changes, I'll upload the "development" kernel sources soon.

Changes compared to the kernel included in v1.5 are:
- CDC ACM driver added (for Pixhawk USB port)
- BCM2385 I2C and Toshiba TC358743 driver added (for B101 testing, see call for beta testers here: https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=120702&sid=675f53e3bcb61651abaca167c187ce7c&start=250)
- RTL8192CU driver added (not much tested, works only for RX as bitrate setting doesn't seem to work)
- Atheros medium access parameter THRESH62 lowered from 28 to 16 (should improve R/C link)
- Atheros noise floor calibration enabled (may reduce badblocks on busy channels)
- support for 4 Atheros RX cards and 6 Ralink RX cards (not much tested)

This kernel does not have the mcs radiotap bitrate patch applied, if you want to use it with Ralink cards you need to apply the patch first!

Add `dwc_otg.int_ep_interval_min=2` to /boot/cmdline.txt to enable support for 4 Atheros RX cards.

For I2C B101 support, add this to /booc/config.txt:
```
dtparam=i2c1=on
dtparam=i2c_arm=on
```

and this to /etc/modules:
```
i2c-bcm2708
i2c-dev
tc358743
```

rodizio1, 2017-07-06:
Please note, I'm currently testing 4.9.28 from the 2017-06-21 Raspbian release, so far it seems that the usb jitter problem is fixed with that kernel. If it turns out to be stable, I'll make some patches and continue with that kernel, so it probably doesn't make too much sense right now to base something on the 4.4.11 kernel.

rodizio1, 2017-07-09:
Well, 4.9.28 works fine on a Pi2 and Pi3, on a Pi1A+ it crashes though :(

Now trying 4.9.35 from Raspbian 2017-07-05 ...

rodizio1, 2017-07-10:
4.9.35 also crashes on the Pi1A+. Both with Atheros and Ralink cards.

bortek, 2017-07-10:
I guess B102 should work as well since they use the same chips?

rodizio1, 2017-07-10:
I'd assume that also, but not sure.

bortek, 2017-07-10:
We shall soon see. I have ordered the board now.

ChristianBueschel, 2017-07-11:
Do you plan to use the B102 with a PiZero since it has also the  22 pin FPC connector with 0.5m pitch?
I was wondering wether that will work for a while now. I have bad experiences connecting a B101 to a PiZero by using one of these: http://www.ebay.com/itm/Camera-Adapter-Board-15Pin-1-0mm-to-22Pin-0-5mm-Pitch-for-Raspberry-Pi-Zero-1-3-/272338708346

bortek, 2017-07-11:
I plan primarily to use with zero. If it wont work then pi3. Not sure which cable though, perhaps the brown pi zero camera cable works. Need to check pinouts. 

But the product page says it comes with cable "ships with 50mm 22 pin FFC cable (0.5mm pitch) and a cable set for the 10 pin connector" so I hope I can use it. 

Otherwise there are these ones available 

http://m.ebay.com/itm/FFC-Flat-Cable-Flex-Ribbon-22-PIN-0-5-mm-Pitch-30-cm-Same-Side-AWM-20624-80C-60V-/181811014086

ChristianBueschel, 2017-07-11:
I've got some experience trying to connect a B101 to a Zero. The B101 needs a 15pin cable with 1mm pitch and the pins on the same side to be connected to a Pi (not Zero!). A Pi Camera Module needs a 15pin 1.0mm cable with the contacts reversed.  15pin 1.0mm is the type of connector you can find on all the Pi's EXCEPT the Zero. 

The Zero has the 22pin, 0.5mm pitch connector that the B102 has as well. On the Pi Zero the connector is mounted in reverse. So in order to use a Camera Module you need a cable 22pin, 0.5mm <> 15pin 1.0mm with the pins on the same side which is the brown cable you are referring to.

In order to connect a B101 to the Zero you would need a 22pin, 0.5mm <> 15pin 1.0mm with the contacts reversed. As there are no such cables available, you'd need the Adapter Board i posted in my previous comment here or you solder such a cable yourself, as shown here:  https://www.rcgroups.com/forums/showpost.php?s=d807a407c592be911f7025cfec5aa01f&p=36631108&postcount=2007

However you want to connect a B102 to a zero. Based on my experience that the HDMI Boards always need the opposite cable to the Camera Modules, I expect you to need a 22pin 0.5mm pitch cable with the contacts on the opposite sides. 

Such cables can be found here: https://auvidea.com/product/42110/
or http://www.ebay.de/itm/FFC-B-22Pin-0-5Pitch-10cm-Flachbandkabel-Flat-Flex-Cable-Ribbon-AWM-Flachkabel-/272404934534?hash=item3f6c996f86:g:WKQAAOSwZJBX99k4

Might be that I'm wrong on this point and the same-side cable that comes with the B102 will work. Interested in hearing your experiences once you receive the B102!

Using the B102 with a regular Pi3 could turn out to be difficult as you would need a cable with 15pin 1.0pitch on the one end and 22pin 0.5pitch on the other end with the contacts on the opposite sides. Same cable needed to connect B101 to the Zero. As I said nobody seems to sell these kind of cables and I've had bad experiences with the Adapter Boards that are sold on eBay or at Auvidea: https://auvidea.com/product/38237/

seeul8er, 2017-07-11:
Makefile in provided development Kernel says 4.4.11.
Did expect 4.9.28 or am I missing something :grimacing:

rodizio1, 2017-08-07:
Little update regarding the 4.9.35 kernel I've been testing: The crash on the Pi1A+ was actually caused by too much overclocking, it seems the A+ doesn't like 1000MHz CPU clock, also not with the 4.4 Kernel, but with the 4.4 kernel, it takes 12-24 hours until it crashes, sometimes even more. While with the 4.9 kernel it crashes after half an hour or so. Strange, seems like the 4.9 kernel just exposed the issue more.

Anyway, with 950 it's running rock-solid straight for a week and I will release 1.6 with 900Mhz (another 50Mhz lower) to be on the safe side.

What's really nice about the 4.9 kernel is, that jitter is really lower and the multi-tasking and scheduling seems to work better and more efficient. In conjunction with the new raw socket tx, this lowers load significantly and will allow for around 8Mbit (instead of 6) videobitrate on the Pi0/1 or Odroid-W, even with the lowered CPU clock.

az0uz, 2017-08-07:
Good to know, I have been playing with 4.9.28 for a while now, with just the wifibroadcast tx/rx script,
and a AWUS052NH.
I have noticed that if I set up a channel without continuous data transmissions, let say I do a `cat /dev/stdin | sudo tx ...`, the data is buffered somewhere and it needs a certain amount before sending a wifi packet even with `-m 1` exemple:
This is not a real problem in the air transmitter (video data throughput is high enough so this is not noticeable) but for uplink (Joystick or camera control) it can add a lot of latency.
A workaround I did is to use `select` when reading stdin in `tx.c`script to send empty packets (header only) when no data is read after a few ms. Using this patch a messaging script using `cat /dev/stdin` works perfectly.

Another suggestion for futur version would be to use the raspicam v4l2 driver instead of raspivid so you can change camera settings while streaming (if you have any kind of uplink) and could make the script compatible with other usb cameras.
for this to work, enable bcm2835 v4l2 module:
`echo "bcm2835-v4l2" | sudo tee -a /etc/modules`
then reboot,
and stream the raspicam:
`
v4l2-ctl -d /dev/video0 \
                     --set-fmt-video=width=1280,height=720,pixelformat='H264' \
                     -p 44 \
                     --set-ctrl video_bitrate=3000000,repeat_sequence_header=1,h264_i_frame_period=44,white_balance_auto_preset=5
cat /dev/video0 | tx ...
`
and you could change the exposure while streaming (dji like)
`
v4l2-ctl -d /dev/video0 --set-ctrl auto_exposure_bias=14
`
or even put manual exposure time, iso. rotation, filters etc...

rodizio1, 2017-08-20:
My R/C tx code runs fine with very low latency, it's not using wifibroadcast, it injects single packets that are as short as possible.

The problem you are having could be due to the terminal buffering maybe. Or because of the wifibroadcast "FEC window". Did you use "-b 1 -r 0" for your tests?

Regarding v4l: I've tried it out, uses more CPU compared to raspivid, so not really useable on the Pi0/1.

Controlling the cam parameters from the ground would be nice though. I've done some testing with Python Picamera some time ago (also uses more CPU though ...), bitrate can be set on-the-fly without any interruption of the video. Together with wifi datarate selection via radiotap header, that would allow for completely dynamic bitrates, when signal is strong it go as high as 15-20mbit, when flying far away it could be reduced to 2Mbit.