rodizio1/EZ-WifiBroadcast #23: Build guide for developers

digitalentity, 2017-06-22:
Project would probably benefit if there would be a howto about building own firmware images - this way developers may start coming to this project and helping improve it.

rodizio1, 2017-06-24:
It's just a Raspbian Lite image where I make changes and then pull an image of it with dd.

Making a build script wouldn't make much sense, requires too much work and testing and will probably break with changes to the upstream distribution. Also, changes to the upstream distribution will break Wifibroadcast, like for example newer kernels give badblocks because of usb timing issues. 

Best thing to do would be to use Buildroot. Much smaller and leaner, most of the stuff included in Raspbian is not needed anyway. And it's better maintainable and 'buildable'.

But I don't that much time to do that, I'd rather spend it on improving some other things first.

If you or anybody wants to start a buildroot wifibroadcast distribution based on my improvements, I'd be happy to contribute and help out with knowledge.

digitalentity, 2017-06-25:
@rodizio1 thanks for the info. Yes, buildroot will certainly be the best thing. For the time being I'm just digging into the code, so Raspbian is probably enough to tinker here and there a bit.

amin005, 2017-07-14:
I may sound like a troll but please add sapport to allwinners processors like H3 and H5 there are so many boards run at 1.2 Ghz with CSI camers out there  with half Raspberry  size and price  please give orange pi and orange pi zero  a look  it has a  Raspbian image  . 

https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20170714064654&SearchText=orange+pi+zero+h5

amin005, 2017-07-14:
I could ship you full set if you ready to give it a shot

az0uz, 2017-07-14:
What distribution runs on those guys ?

amin005, 2017-07-15:
http://www.orangepi.org/downloadresources/

rodizio1, 2017-07-17:
It's not 'adding' a little here and there, it's basically 'doing it all over again'. 

These clones have a different boot mechanism and firmware, it would require to make another image. Next problem is, that the Pi clones come in many different variations and models, and are not a 'stable' platform like the Pi, where you can be relatively sure that the models stay like they are.

Then you need applications for:
- Hardware accelerated low-latency h264 encoding of the camera video
- Hardware accelerated low-latency h264 playback
- Hardware accelerated rendering for the OSD

I don't know what the situation is today but about 1 1/2 years ago when I started with EZ-Wifibroadcast, the above things did not exist on the Pi Clones. Maybe that has changed now. But still, I won't be investing time into this.

amin005, 2017-07-17:
why you call them clones anyway !
these are completely different  single board computers  and they are well designed not a clone or what so ever 
as I can clearly see here you put some sort of imaginary obstacles front of you  as you judge this board based on  what you think they . 
you keep spending time on RPI  it is bulky and not cheap not suitable for most of fpv application

az0uz, 2017-07-17:
I am working on a debian package based on systemd and splitting every functionalities in different scripts / modules.
even if openMax or any other hardware encoder wouldn't be available on Orange Pi, it could work with a H264 encoded usb camera. Play back doesn't seems as hard to find, and I think openVG is available on the orange pi.

closed 2017-10-20 by rodizio1