rodizio1/EZ-WifiBroadcast #26: About FPS Option in Config File

jiejiecn, 2017-06-30:
Q1: In the config file (line 61)
_Choose between 30, 48, 59.9
FPS = XX_

What if I set FPS = 35 or 45 or 50 ? 

Q2: Why it's 59.9 rather than 60?

rodizio1, 2017-07-02:
There are three differen hello_video.bin (the program that displays the video) included. 

1. hello_video.bin.48 by mmormota, useable from 48 to 59.9 fps (optimized for smooth playout)
2. hello_video.bin.30 by mmormota, useable for 30fps (optimized for smooth playout)
3. hello_video.bin.240 original by befi, useable for all other fps (but not perfectly smoot playout)

So if you choose either 30fps, or something between 48 and 59.9fps, mmormota's hello_video.bin will be used and you get smoother video, for all other fps settings the "standard" hello_video.bin will be used. 

Not sure exactly why it's 59.9 fps max for mmormota's hello_video.bin, but there is some technical reason.

jiejiecn, 2017-07-05:
@rodizio1  Thanks. I'll look into that project.

closed 2017-07-05 by jiejiecn