rodizio1/EZ-WifiBroadcast #27: My Tips on Low Latency

jiejiecn, 2017-06-30:
Recently I achieved a quite good latency, around 80~100ms under 1280*720 / 48FPS / 4Mbps, using v1.5 with two RPi2

Here're some snapshots with my own program, the number increased by 10 every 10ms. (This is because the response time of LCD display on my laptop and desktop is around 10ms. So with 10ms interval, I can take a clear photo to measure the latency.)
[image]
[image]
[image]
[image]

On Tx side it's RPi2 with WN722N and on Rx side there're two USB AR9271 module for diversity.
[image]
[image]

The AR9271 is really really a good choice for this purpose. It works under 2.3G~2.4G, I choose 2372MHz which is near 2.4G for antenna and also the channel is clean enough.

Secondly, the USB AR9271 module on Rx side is only $3 and you can find small module with 25dbm RF output for Tx side for only $10

[image]

Here're some tips:

1. High speed TF card.

This is tricky but it's true! I use SanDisk Ultra 16G Class 10 A1 on both Tx and Rx side. It's much better than Class 4 TF card.

2. GPU / RAM overclock

The video is encoded by GPU on the SoC. So less latency with higher frequency on GPU.
I put GPU and RAM overlock in config.txt
_gpu_freq=500
sdram_freq=500_

3. High FPS

Set the FPS 48 or 60, the higher FPS the less latency. 

4. Camera Sensor Mode

I use RPi Cam v2 and according to the IMX219 specs, set the "-md 6" to force sensor mode to provide high FPS output. 
[image]

And someone did the test on this.
[image]

_EXTRAPARAMS="-n -fl -ih -vf -pf high -md 6 -ISO 400 -ex sports"_

pilotnbr1, 2017-07-15:
That's pretty good results!

bortek, 2017-07-21:
Thats a pretty interesting and and good description. I have updated Wiki page (https://github.com/bortek/EZ-WifiBroadcast/wiki/Tips-and-tricks) and reffered to this Issue for those who want to tune their systems for lower latency. I therefore will close this issue for now.

closed 2017-07-21 by bortek

az0uz, 2017-07-31:
Quick remark on the settings given here, I think most of us wouldn't want to give up half fov for a few ms lower latency, so I wouldn't advise `-md 6` but prefer `-md 4` or `-md 5`