rodizio1/EZ-WifiBroadcast #30: USART on Pi Zero W?

Toni-42, 2017-07-12:
Hello,

I  just tried the new instructions for Pi Zero W (replace the start_x.elf file from 2017-03-03 firmware). Pi Zero starts and works fine. But the thing is, the  LTM protocoll doesn't work on Pi Zero W. If I change to the Pi 3B on TX-side (and original 1.5 Image Data on the SD-Card with same settings like Baud in wifibroadcast-1.txt) the LTM protocol works ok.

Is this problem known?

Greetings Toni

rodizio1, 2017-07-13:
So far I haven't heard of such a problem. Can you post the wifibroadcast-1.txt please?

Toni-42, 2017-07-14:
No problem:
[wifibroadcast-1.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1147478/wifibroadcast-1.txt)

rodizio1, 2017-07-15:
That looks like it's correctly configured. Hmm, not sure what the reason could be. Have you double checked the wiring and made sure that your flight control's serial port uses 3.3V? Maybe the Pi0 serialport has an issue?

Toni-42, 2017-07-16:
Hello
the question is (sorry for that maybe very stupid question), has anyone ever tried the complete functionality including serialport (OSD) with the Pi Zero W? If so, then the error is definitely on my side.

My FC has 3.3V (voltage divider from 5V->3.3V) I have measured with the Oszi. I can also see the data via USB/Serial Converter on the PC. As soon as I plug (change form Pi Zero) to the PI 3 B everything is ok.

This may have something to do with the start_x.elf file? I mean, the serial interface is switched off in this file...? What is the function of this file..?

rodizio1, 2017-07-16:
I haven't seen somebody explicitly reporting back that the serialport works with the Pi0 W, I just assumed that, since there are Pi0 W users and nobody complained so far. Could of course be, that nobody has used the serialport yet on the Zero W.

The start_x.elf is part of the firmware, the Zero W needs a newer firmware to work.

rodizio1, 2017-07-17:
Can you connect a monitor and keyboard to the Pi0? If yes, you could check if the serialport is there and if it's working. You can login by switching to tty12 with alt-F12 and then type 'cat /dev/serial0', that should print out the raw telemetry data to the screen (a lot of 'garbage').

Toni-42, 2017-07-18:
Ok, I'll try it on this weekend.

Toni-42, 2017-07-24:
Hello rodizio1,
I have tried your proposal and i noticed that the serial port didn´t send any sign.
Just to be sure: after typing "cat /dev/serial0" the answer on pi was "cat: /dev/serial0: No such file or directory".
Toni

seeul8er, 2017-07-24:
On my project with a RPi3 I am using `/dev/ttyAMA0` to get serial data from UART port (no bt-overlay). `/dev/serial0` does not work for me either

rodizio1, 2017-07-25:
> Just to be sure: after typing "cat /dev/serial0" the answer on pi was "cat: /dev/serial0: No such file or directory".

That shouldn't be. /dev/serial0 should definitely be there. Can you configure a wrong frequency (like 3000) on the TX so that it throws an errorlog.txt and post it here? Maybe the dmesg output in there gives some hint.

> On my project with a RPi3 I am using /dev/ttyAMA0 to get serial data from UART port (no-bt-overlay). never tried /dev/serial0 though

This was quite a confusion, if I remember correctly, at one point they changed it because of the Bluetooth introduced on the Pi3. 
However, then, they introduced /dev/serial0, which should always point to the correct serialport, not depending on Pi model, that's the recommended device file.

@Toni-42 
Not sure right now if this is some kind of strange problem with the firmware. You could maybe try /dev/ttyAMA0, /dev/ttyS0 or /dev/serial1 and see if one of them works.

Another option would be physically broken serialport, you could test that with a standard Raspbian distribution image (requires some configuration work though, the serial console must be disabled if I remember correctly)

Toni-42, 2017-07-26:
Ok, here is the errorlog.txt with the wrong FREQ=3000
[errorlog.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1177953/errorlog.txt)

I tryed also another inputs, everthing failed, see the screenshot.
[image]

rodizio1, 2017-07-27:
Sorry I should've been more clear about the commands, it should've been:

"cat /dev/ttyAMA0", not just "/dev/ttyAMA0"

In the errorlog.txt I can see a serialport named /dev/ttyAMA0 is being detected, so I think that one should work.

Toni-42, 2017-07-27:
OK, with "cat /dev/ttyAMA0" I got no answer from PI on the screen (I think this is what you expected) and no sign via serial port. What does that mean, the serial port is dead? This is implausible that inside the chip the serial port has an issue. May be a circuit path is damaged.

It would be great to know that another user have no problems with PI Zero W serial port and 1.5 Version.

rodizio1, 2017-07-30:
If you get nothing back, atleast the device file is there. If you have your FC connected while issuing the "cat /dev/ttyAMA0" command you should see some 'garbage' (the serial data from the FC) being printed out. 

Apart from trying setting /dev/ttyAMA0 in wifibroadcast-1.txt I have no more ideas.

I will hopefully soon release EZ-Wifibroadcast 1.6 Beta, this release will include the firmware and kernel from the latest 2917-07-05 Raspbian. If it is some kind of firmware or kernel issue with the Pi0W it will hopefully be fixed then.

az0uz, 2017-08-02:
Maybe you could try the following to kill any process already using the port.
replace 9600 with the baudrate of your telemetry (9600 seems to be the default in cleanflight)

```
sudo fuser -k /dev/ttyAMA0
sudo stty -F /dev/ttyAMA0 raw -echo -echoe -echok crtscts -onlcr -echoctl -echoke 9600
cat /dev/ttyAMA0
```

Toni-42, 2017-08-02:
Hello az0uz,
if i connect a monitor and keyboard to the Pi0 (has only one USB port) an then login by switching with alt-F12 typing all this instructions "sudo..." the Tx-Pi will not work till it has a power reset? Even if i kill a "bad" process, how can i get the result?

az0uz, 2017-08-02:
Do you mean that after executing the given commands, the Pi totally freeze ?

Toni-42, 2017-08-05:
No, what I mean is that the state changes after input / startup. [If the Pi does not find Wifi (because the only one USB Port is connected to the keyboard), then it breaks its work off.]

I hope that with the release 1.6 the developers explicitly check the function (Serial Port->LTM/Mavlink ptotocol) with the Pi zero W.

rodizio1, 2017-08-07:
I don't own a Pi0W, so I cannot test myself unfortunately. But I'll test on a Pi0 and Pi1A+. I've already added some more checks and error messages (if the serialport is there) to upcoming version 1.6, so it should be easier to pinpoint the issue if there is something different with the Zero W.

Have you tried changing the serialport to "/dev/ttyAMA0" in the wifibroadcast-1.txt yet?

Toni-42, 2017-08-07:
Hello rodizio1,

I like this projekt, so i would like to support it. I could buy for you a Pi0W and send it to the address of your choice. Or i could also transfer 15$ via paypal. (I think Pi0W is the best Pi for Tx Side (small and light), so let's test the things properly.)

Regarding the command in wifibroadcast-1.txt i haven't tried it yet. This point was completly unrelized from me. May be tomorow.

Toni-42, 2017-08-08:
I have changed in the wifibroadcast-1.txt "/dev/ttyAMA0" instead of "/dev/serial0".
Unfortunately without success.

Here is the whole file.
[wifibroadcast-1.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1209382/wifibroadcast-1.txt)

rodizio1, 2017-09-08:
Sorry for the late reply, I asked on the rcgroups forum if somebody has problems with the serialport and the Zero W, nobody answered.

But now there is somebody on another forum who also cannot get it to work with a Zero W, but it works for him with a Pi1.

Probably a bug then, caused either by firmware/kernel or overlay incompatabilities. Should work in v1.6, as that version will have kernel/firmware/overlays from the same Raspbian release.

rodizio1, 2017-10-20:
Should work in 1.6 RC1 and following, closed

closed 2017-10-20 by rodizio1