rodizio1/EZ-WifiBroadcast #31: Telemetry data corruption issue (and fix)

rodizio1, 2017-07-15:
User r0ck3t identified a bug that had gone un-noticed since the beginning of wifibroadcast. The serialport options caused some characters to be replaced (CR/LF), which leads to corrupted telemetry messages if these characters happen to be in the telemetry messages.

This will be fixed in the next release. To fix this for yourself in the meantime you can login to the linux system on the TX (see the wiki on how to do that) and make a change to the .profile script.

When logged in:
- type "rw" to make the filesystem writeable
- type "mcedit .profile" or "nano .profile" (whichever editor you may prefer)

Find this line:
`FC_TELEMETRY_STTY_OPTIONS="-imaxbel -opost -isig -icanon -echo -echoe -ixoff -ixon"`

and change it into this:
`FC_TELEMETRY_STTY_OPTIONS="-icrnl -ocrnl -imaxbel -opost -isig -icanon -echo -echoe -ixoff -ixon"`

- Save the file and exit the editor.
- type "halt", wait a few seconds and then power-off the Pi

rodizio1, 2017-08-20:
Update:

More tests have shown that these settings should improve things:

TELEMETRY_BLOCKS=1
TELEMETRY_FECS=1
TELEMETRY_BLOCKLENGTH=267
TELEMETRY_MIN_BLOCKLENGTH=24
FC_TELEMETRY_STTY_OPTIONS="-icrnl -ocrnl -imaxbel -opost -isig -icanon -echo -echoe -ixoff -ixon"

bortek, 2017-08-20:
I can update wiki with these but will they go into next release?

rodizio1, 2017-08-20:
Yes, I'm currently writing a new telemetry tx that should handle this much butter than the current wifibroadcast tx in V1.5 

The above changes should make it better hopefully until v1.6 is ready.

ivanskj, 2017-09-16:
Hi @rodizio1 , Any estimate on 1.6 release date?

geofrancis, 2017-10-05:
I have just got v1.5 up and running with a pair of pi3s,

telemetry would kind of work, I got a few random sensors.
changed the settings you recommended made the telemetry link work great, i am now getting around 1k/s bandwith at 57600 and everything looks like its working, its loading parameters etc.

looking forward to mavlink rc but its a shame its limited to 8 channels.

rodizio1, 2017-10-21:
thanks for reporting back. V1.6 includes a new telemetry tx and rx, so I close this issue.

closed 2017-10-21 by rodizio1

geofrancis, 2018-05-23:
1.6 has issues still, this is still the only way i get working mavlink telemetry.