rodizio1/EZ-WifiBroadcast #32: Rx:  Error setting monitor mode on failed!

tobyfarrar, 2017-07-25:
I just want to start by saying that this is a fantastic project! I always hoped that this was possible, but didn't have the skills to make it happen. So a huge thank you to you for making it happen.

When I first booted up both Pi's everything worked perfectly. I ran the system a couple of times, again perfectly. Today I booted up them both and the Rx Pi came up with the "Error: setting monitor mode on failed!" (I've added a picture of the screen here)
[image]

The only thing I had changed was to enable the wifi hotspot and change the SSID and password. I have re-imaged the SD card with Etcher, but this error keeps happening. Any help would be greatly appreciated.

Thanks

rodizio1, 2017-07-25:
Looks like either power or wiring problems (then all kinds of strange things can happen ...) or a syntax error or some other problem with wifibroadcast.txt. 

Could it be that you have some GPIOs connected to something? Or that they are broken/malfunctioning? Could be that it's looking for a wifibroadcast-8.txt (or some other configuration profile number, see the Howto section of the wiki).

Can you please post the errorlog.txt from the sdcard here?

tobyfarrar, 2017-07-26:
[errorlog.txt](https://github.com/bortek/EZ-WifiBroadcast/files/1175825/errorlog.txt)
[image]

It is a Pi 3 Model B and is powered from the official 2.5A 5.1v power supply via the micro USB port. There is nothing attached to the GPIOs. All that is connected is the power, HDMI to the monitor and the CSL-300 adapter into one of the USB ports. The strange thing is that it was working fine the first few times I powered it up. Would you suggest that I power it from a better supply (I have a 5A SBEC that I will be using eventually in the field)?

tobyfarrar, 2017-07-26:
Update:

I had a spare RPi 3, so I swapped the SD card into it and connected it up and it booted up perfectly. I put the card back in the original one and it has the same error. I guess therefore it is a problem with the board?

rodizio1, 2017-07-27:
Yes, this looks like one or more GPIOs is broken which causes a different wifibroadcast.txt to be selected (which doesn't exist). See here for an explanation of that feature: https://github.com/bortek/EZ-WifiBroadcast/wiki/How-to's#using-configuration-profiles

A quick workaround would be to copy over the wifibroadcast-1.txt to wifibrocast-2.txt and so on up to wifibroadcast-8.txt.

Better way would be completely removing that feature, I can look up how to do that later.

tobyfarrar, 2017-07-27:
I did your workaround and that has sorted it out. Thank you so much for your help. I will test the GPIOs on this board and see what that problem is, but at least for now it is fully functional.

closed 2017-07-27 by tobyfarrar