rodizio1/EZ-WifiBroadcast #35: Odroid-W support,   camera connect by HDMI to CSI.

AAV514, 2017-08-06:
Hi,  
I have some questions.

Q1.
I read this notes, 
TX: The CPUs on the Raspberry Pi 1 and Pi Zero are more or less maxxed out with standard settings (6Mbit bitrate, 8/4/1024 FEC). 
Two TX dongles, higher resolution/bitrate, or more error correction or smaller packet sizes will most likely not work. 
Even if it seems to be working on first look, it can happen that latency suddenly raises if the CPU is loaded to much (in situations with high bitrate, 
like fast scene changes). Although many people successfully use a Pi Zero as a TX, if space and weight is no issue, consider using a Pi2 or 3 for TX.

RX: Raspberry Pi1 and Pi Zero are not supported anymore from version 1.3 on. Use a Pi2 or Pi3.

Odroid-W use same SoC BCM2835  as Raspberry Pi1,    
then, I can't use Odroid-W as a RX, and better not to use it as a TX? 

Q2. 
According to guide, EZ-Wifi Broadcast use Pi V1 cam or V2 cam.
but I want to use a better camera for Wifi Broadcast.
My camera has HDMI output port on it.
if I use a HDMI to CSI converter like this,

B101 HDMI to CSI-2 bridge 
https://auvidea.com/product/70501/

Can I connect my camera on RPi and can compose EZ-Wifi Broadcast system?

rodizio1, 2017-08-07:
Odroid-W should work fine with stock settings, I think the majority is actually using a Pi0 as TX. It's just that maximum bitrate and FEC settings are limited by it's CPU, so you cannot go much higher with bitrate and/or FEC. 

That being said, I'm currently experimenting with kernel 4.9 and a wbc tx using raw sockets, it looks like this combination is more efficient, so that bitrates of about 7-8Mbit should be possible with the Zero or Odroid-W in the future with upcoming version 1.6.

Since you already have the Odroid, I'd just get a Pi2 or Pi3 as RX and use that. If you feel you want higher bitrate or FEC settings later you can still upgrade.

AAV514, 2017-08-07:
Really thank you for your comment.

closed 2017-08-08 by AAV514