rodizio1/EZ-WifiBroadcast #36: External amplifier  &  MIMO WIFI card.

AAV514, 2017-08-08:
Hi,
I have a question, 
If I attach a amplifier on multi antenna MIMO support WIFI card,
I should have to attach amplifiers  on all antenna port?

for example, 
2 antena port WIFI => need 2 amplifiers
3 antena port WIFI => need 3 amplifiers

AlexandreLicinio, 2017-08-08:
hi, it depends on what you expect but consider you have to amplify the TX
part rather than RX.
but as you said in mimo all antenna do TX + RX (or watch if you schematic
of antennas).
The better amplifier you can have is ANTENNA (if you could) for TX and RX.
This one is bi-directional >
http://www.alfa.com.tw/products_show.php?pc=43&ps=123

-- 

*Alexandre Licinio*

*E**KLA INGENIERIE *
*Broadcast engineering*
*www.ekla.tv <http://www.ekla.tv>*

*French Partner of Open Broadcast Systems Ltd. *
*422 10 Bits Encoders. Decoders. Cloud Contribution*
*www.obe.tv <http://www.obe.tv>*

alexandre.licinio@ekla.fr

+33 (0)6 76 10 28 41 / +33 (0)9 81 82 09 08
50 Avenue Lacassagne
69003 Lyon - France

rodizio1, 2017-08-12:
I think it won't work properly with an amp only on one antenna port. At a certain distance, the signal from the antenna without amp will not arrive at the receiver anymore due to the lower power output, which makes it ineffective. Also, I'm not sure if the STBC still works properly, could be that small timing or whatever signal differences introduced by the amp on only one port makes it less effective or interfere with itself.

AAV514, 2017-08-12:
Yes, I think so too.
That's why I'm worried.

rodizio1, 2017-08-12:
A forum member on rcgroups is using those "2W" XQ-02A Phantom amp modules with success. Measured power output is about 400-500mW and they are not too expensive.

This one for example is 12$:
https://m.banggood.com/YXS-XQ-02A-2_4G-2W-Bi-directional-Signal-Amplification-Module-p-1059396.html

AAV514, 2017-08-13:
if I use  a two antenna ports WIFI card, had better to attach AMP on all two antenna ports?

AlexandreLicinio, 2017-08-13:
Since card are mimo so antenna works both tx and rx, yes you should attach
an amp on each port.

closed 2017-09-05 by AAV514