rodizio1/EZ-WifiBroadcast #37: Question about diversity WIFI compose on receiver side

AAV514, 2017-08-13:
Hi, 
I have some question.

1.  I plan to use Pi-ZERO as a transmitter, because of it's small size. 
     In this case, I can't use 2 WIFI  diversity transmit Function?

2.  I read this,  
     3 wifi sticks receive diversity support for Atheros
     5 wifi sticks receive diversity support for Ralink
     then, how about MediaTek MT7601?  Is it same to Ralink??

3.   I plan to use RPi 3 as a receiver.  I want to make a 2~5 WIFI diversity receiver. 
      but  I don't want to use  USB port type WIFI stick, because it is too bulky.
      Can I compose a 2~5 WIFI diversity receiver without use of USB port connection?
      I mean, can I connect 2~5 WIFI cards with  RPi 3 by use only the GPIO pin?

rodizio1, 2017-08-18:
1. Yes, MT7601 uses the Ralink driver, so 5 cards should work.

2. Not possible. But you can de-solder all the USB plugs and sockets on the USB dongles and the Pi (for the TX, this is mandatory anyway ...) and solder wires directly, this makes it a lot smaller. 

3. Two TX dongles on the Pi0 or Pi1 is not possible, not enough CPU power to do that.

4. You can use either one dongle on the TX (one frequency), or use the dual-TX function with two dongles on the TX (on two different frequencies). Two TX dongles on the same frequency won't work.

The dual-TX function with two dongles works only with Ralink (and the MT7601) dongles both on TX and RX, Atheros gives timing issues and thus badblocks.

One TX dongle combined with multiple RX dongles is the default, there is nothing that needs to be setup for that, it just works.

AAV514, 2017-08-18:
Really thanks a lot~!!
every question has been resolved

closed 2017-09-05 by AAV514