rodizio1/EZ-WifiBroadcast #38: use the MT7610 instead of MT7601.

AAV514, 2017-08-14:
I'm trying to use MT7610 instead of MT7601.
because I want to use 5GHz WIFI connection.
EZ-WifiBroadcast can support this chipset?

MT7601U 
2.4
bgn 
1x1:1 

MT7610U 
2.4 / 5
abgn+ac
1x1:1

rodizio1, 2017-08-18:
MT7610U will not work I think.

AAV514, 2017-08-18:
Shall I try it??
Is it worth to challenge?

AAV514, 2017-08-18:
MT7610

https://www.alibaba.com/product-detail/11ac-wifi-module-433Mbps-MT7610-for_60610863176.html?spm=a2700.8443308.0.0.Q2lkoH

support  5GHz - No interference with 2.4GHz Transmitter.
support 1 external antenna port - Esay to connect WIFI amplifier.
compact size - 42 * 19mm 
relatively cheap & new chipset - $7 for each, easy to obtain.

rodizio1, 2017-08-18:
According to this thread https://forums.fedoraforum.org/showthread.php?t=309580 it doesn't work with the rt2800usb driver. So, no, won't work.

AAV514, 2017-08-19:
Oh, I know rt2800usb driver can't support  MT7610.

but MT7610 works on your link page 2.

also maybe works on Raspbian
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=174784&p=1115732&hilit=mt7610#p1115732

AAV514, 2017-08-19:
How about  QCA9377 ??

rodizio1, 2017-08-21:
Only the chipsets listed in the Wiki work. I

I think for the qca9377 there is no driver at all for linux.

V1.6 will have drivers for Realtek 8812/14 and others, but I still wouldn't recommend buying them as I have done zero testing (don't own any yet), so all kind of things could be buggy or not work at all. If you have some Realtek cards laying around anyway, you might want to give it a try.

AAV514, 2017-08-22:
Thank you for your reply.
I'll wait  v1.6

qca9377
https://wikidevi.com/wiki/List_of_Wi-Fi_Device_IDs_in_Linux
https://wikidevi.com/wiki/Ath10k
https://wireless.wiki.kernel.org/en/users/drivers

packet injection not supported yet
but maybe supported Sooner or later

please support 5.8GHz,  1x1:1 card. 
MT7610
QCA9377
RTL8811AU 
I'm really want it.

closed 2017-09-05 by AAV514