rodizio1/EZ-WifiBroadcast #39: Mavlink AES encrypting using mavcrypt.py

wiiverueckt, 2017-08-19:
Would it be possible to encrypt the mavlink with this pyton scipt?

https://github.com/Blastitt/mavcrypt/blob/master/mavcrypt.py

An unencrypted mavlink is really a serurity issue.
Unencrypted video only a privacy issue.

seeul8er, 2017-08-19:
Disclaimer: I am no expert on the subject but I also made my thoughts around something like this

The effectiveness of something like an encryption depends on what you want to achieve.
If you want to hinder someone from reading the messages/traffic then it totally makes sense to encrypt.
However if your main goal is to make sure noone manipulates the connection (like sending RC signals or any other harmful commands) it is not very effective in my opinion.

**The problem is that neither the drone nor the groundstation can verify the origin of a received packet.**
An attacker would not be able to read the traffic however he could go and just listen for some frames and then send these (still encrypted) frames over and over again. The drone/groundstation would always accept these frames as they are valid. That way he can easily hijack the connection without knowing the encryption password.
The only way (I can think of) of making sure this does not work out for an attacker is to put something like a sequence number or time stamp (something that increases) into each packet. The receiver would then have to check if that number is somehow valid. However the standard packetloss and the packetloss when the reception goes down to zero for a few seconds can make this task of verifying a sequence number very hard to impossible.

rodizio1, 2017-08-20:
It's surely possible to add that. Not sure about CPU utilization though, probably too much for a Pi0/1.

I have no immediate plans to add encryption to EZ-Wifibroadcast, not worth the hassle at the moment, nobody will target it with such a small userbase when there are hundreds of thousands vulnerable drones like Parrot for example.

bortek, 2017-08-20:
I agree with the last comment even though this is a nice to have feature. Something to consider in the future.

rodizio1, 2017-10-21:
will close this as I don't see this happening in the near future. If somebody contributes code, I'll be happy to integrate it though.

closed 2017-10-21 by rodizio1