rodizio1/EZ-WifiBroadcast #40: .profile TTY options

terial, 2017-08-29:
Hello,

I am making some changes to the .profile and making a shell script out of it. at the end of .profile  i see there is a case statement for each tty1,tty2,tty3.... etc. I can not figure out where these are being defined so the case statement know what to do. Any pointers would be greatly appreciated.

Thanks.

rodizio1, 2017-08-29:
These are the linux TTYs, they are not defined in the .profile. The case basically makes different things run on different ttys.

You can access them the usual linux way with alt-F1 to alt-F12 when you have a keyboard connected.

terial, 2017-08-29:
Ok that now it's a bit clearer. Could the same thing also be done with screen? Or is tty the only way to route data, like from the cmanode link to wifubroad cast?

rodizio1, 2017-08-31:
These are two different things. The cmavnode mavlink data uses virtual serialports (via socat). This is for getting the mavlink data into cmavnode (which only supports serialports, so I had to emulat them).

The ttys for the different screens are just a way to have different stuff running in parallel and also giving the user a chance to see what's going on by switching ttys with alt-F...

Could probably also be done with screen or with bash functions being put in the background, there are many ways.

closed 2017-10-20 by rodizio1