rodizio1/EZ-WifiBroadcast #41: OSD very laggy, sometimes non-responsive

terial, 2017-08-30:
Hello,

I have two RPI 3 with the latest image, the RX has the 7" LCD attached. I have a pixhawk connected and video works great and I see that in the left corner telemetry is being received and decoded but the OSD horizon , compass, etc... are very laggy or just do not respond to any changes in the pixhawk's attitude. Originally I had the baud rate at 921600 but have switched it to the default rate but still no luck. Any suggestions?

rodizio1, 2017-08-31:
Do you have all settings (except frequency) on default settings and a non-modified image?
In downstream direction, telemetry should work fine. There can be small 'stutters' sometimes, but what you describe sounds worse.

Do you get high packetloss? Check the 2nd number in the upper left corner, this is the packetloss counter. If it's steadily increasing, you have interference (probably from other wifi networks around you).

One option would be trying another channel, another option would be removing the antennas on the RX dongles and keeping the TX close (this way, the RX cannot hear the interference, but only the TX).

terial, 2017-09-02:
both the tx and rx were on default images. Only settings changed were the frequency and tried various 5Ghz frequencies. It did not seem like i was getting any dropped packets. I can see the counter for the relieved telemetry messages but it shows 0 as being decoded.

terial, 2017-09-03:
In my above post i mis-typed about the second column for telemetry... the last two counters in the upper left in the parenthesis... the number on the left increases fast, the number on the right does increase but very slowly... i take this to mean that it is not decoding the messages as fast as it is receiving them?

rodizio1, 2017-09-03:
Yes, correct. If it's receiving a lot of stuff, but cannot decode everything, it's either getting a lot of mavlink messages it doesn't know about or a lot of 'broken' mavlink messages (that are getting truncated because of packetloss).

Does the second number (the one for the videostream, not in parentheses) stay zero or near zero? If you get packetloss on the videostream, you will likely also have packetloss on the telemetry stream.

Do you have unusual settings on your FC, like many things enabled and maybe too high streamrate? (see the wiki for infos about that).

rodizio1, 2017-09-22:
Did you have a chance to test it in the meantime? If you set your FC according to this: https://github.com/bortek/EZ-WifiBroadcast/wiki/OSD-MAVLink-message-types

And make sure that you don't have high updaterates on the other SR_ parameters, it should work.

That being said, I've written completely new telemetry tx and rx programs which parse the mavlink data to make sure mavlink messages are transmitted properly. Plus I have made some changes and optimizations to the OSD code, in v1.6 the OSD and artificial horizon will be smooth as butter and low-latency.

terial, 2017-09-28:
Yeah I just gave it a try and no change. I have tried to keep all the settings default but maybe I have something messed up. Any idea when the 1.6 code will be released?

rodizio1, 2017-10-21:
1.6RC1 has been released. 1.6RC2 will probably follow today or tomorrow, please re-test with those versions and open a new issue if the problem persists.

closed 2017-10-21 by rodizio1