rodizio1/EZ-WifiBroadcast #43: How to get internal wifi back as wlan0

terial, 2017-09-02:
Hi, i am wanting to get the internal wifi renamed back to wlan0 and get it running dhcp as default. I tried getting the network interfaces file back to default but it seems like wpa_supplicant doesnt run or is not installed? If you had a few tips i would greatly appreciate it!

rodizio1, 2017-09-03:
Sorry I have never really used wpa_supplicant or network manager, I just de-installed them because they would always mess with the interfaces :)

Not sure how to do that. If you install stuff with apt-get, be careful that it doesn't overwrite confif files etc. and break stuff.

I think the best way would be installing wpa_supplicant from source and configuring it manually somewhere in the .profile script, this way apt and network-manager won't mess up everything.

terial, 2017-09-04:
Thanks, I also see that one of the kernel patches also change the name? I got everything sorted out thanks

closed 2017-09-04 by terial