rodizio1/EZ-WifiBroadcast #44: Network interface broadcast power

terial, 2017-09-04:
Just did a kernel build with raspbian jessie 2017-07-05 with the patches. The first time the dwg_otg patch seemed to create issues when building but since I am not using that feature I started from scratch and omitted it. When i run my script to start the tx (not much difference between the script and the .profile used in your image) the Tx-power goes from 20 to 13. The internal wifi stays at 31. I also did not have any .conf files in /etc/modprobe.d. Is this most likely a problem when I built the kernel?

rodizio1, 2017-09-05:
I have never really looked at the ifconfig txpower output, it seems to always show stuff that isn't true (30dbm =1Watt for the internal card, yeah right ...).

I'm not sure right now what default values I've defined, could be that it gives lower values when you don't use the txpower module parameter.

the dwc_otg patch doesn't apply to 4.9.35 because it's already included in that version, I made the patches for 4.9.28 (and then, a few days later, the Raspi Foundation made a new release...).

terial, 2017-09-05:
Ok thanks. Yeah I was a little hesitant on the internal card putting out that much and my Alfa 52 not. Thanks for all your help!

closed 2017-09-05 by terial