rodizio1/EZ-WifiBroadcast #45: connect RTL8811AU on RPi.

AAV514, 2017-09-05:
Hi, 
I have connected a RTL8811AU WIFI cards to RPI 3,  and made it works on 5.8GHz.
I don't know it works well with Wifi Broadcast, but connection itself is not that difficult.  
Would you please support the RTL8811AU cards on the next version of Wifi Broadcast?

closed 2017-09-29 by AAV514

scoquereau, 2017-11-08:
hi,

It seems to need to install a driver for the RTL8811AU. Could you explain which one you have choosen and installation procedure ?. The RTL8811AU chipset is cheap and could be a real solution for wifi broadcast

AAV514, 2017-11-08:
hi, I used this driver .

https://github.com/marczis/rtl8812au

it needs some update & compile 

==================

sudo apt-get install rpi-update
sudo rpi-update
sudo apt-get install git bc libncurses5-dev

sudo wget "https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source" -O /usr/bin/rpi-source
sudo chmod 755 /usr/bin/rpi-source
rpi-source --skip-gcc 
sudo git clone "https://github.com/marczis/rtl8812au

cd rtl8812au

sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile

sudo make clean
sudo make -j4  RTL8811=1 DEBUG=1 

sudo cp 8812au.ko /lib/modules/`uname -r`/kernel/drivers/net/wireless
sudo depmod -a
sudo modprobe 8812au

scoquereau, 2017-11-08:
Thanks a lot, I will try this.

scoquereau, 2017-11-12:
Hi AAV514,

I have compile with your procedure the driver, it seems to work well, and i can see some frame on the RX. But the image seems to freeze and i'm unable to get it works. Could you say me, if you have make change in the  wifibroadcast-1.txt or what is your configuration.

Mine is zero W TX and pi2 RX

Thanks

geofrancis, 2017-11-14:
was there any progress on this ?

scoquereau, 2017-11-26:
Hi francis, sorry but it doesn't work for me with 1.6 RC2, i have no image on the screen.