rodizio1/EZ-WifiBroadcast #46: wifi regulatory zone settings

chrispable, 2017-09-19:
Just a heads up, I discovered this "bug" by accident on a separate project I am working on but originally wanted to use this "distro" as a base. I ended up not doing it but my usecase involved an adapter that defaulted to an unspeced wifi regulatory zone. I tried and tried to use channel 161 on 5ghz to broadcast a packet. It only worked if it could see something already talking on channel 161.

My suggestion would be to make the wifi regulatory zone setting part of your config file. As soon as I set it to US, I was able to use 161 reliably, otherwise the usb command wouldn't even get sent to the adapter!

rodizio1, 2017-09-22:
The regulatory settings should be fine in the image for the supported ath9k_htc and rt2800usb dongles, you can transmit on any channel you want.

If you can only transmit when there is other wifi traffic around, you have the NO-IR flag on that channel. Setting different countries can help, but it will never work for all channels because there is no country in which all channels are allowed.

The channel limitations you are seeing is probably because the driver of the dongle you're using also has country-specific limitations (similar to ath9k, where I have removed the limitations). Or the firmware of the dongle.

What dongle did you use and what country did it default to? If you want, you can send me an errorlog.txt (just configure a completely wrong frequency like 4000 to make it throw an error and write the errorlog.txt to the sdcard), then I can check if the country is not in the db.txt

chrispable, 2017-09-22:
I used an rt2870 in a panda pau09. It defaults to 00unspec. Using iw reg
set US I can use channel 161 which is what I wanted to use.

Without that I could only broadcast on channel 161 if there was a router or
access point that was already talking on that. I no longer have a spare
unit and didn't mean to imply a bug in the software, merely suggesting to
optionally have that setting in the config to run that command. I just
didn't wanted the issue documented so some poor random googler doesn't get
stuck for weeks like I did.

rodizio1, 2017-09-23:
Hmm, that one uses the rt2800usb driver (I assumed you were using some different chipset). Well, then it must be a bug. But it's strange, db.txt contains country 00 and I've also patched the 'hardcoded' stuff in the kernel sources. Not sure where the no-ir flag has been coming from in your case. You were using ez-wifibroadcast 1.5 (or atleast it's kernel)?

chrispable, 2017-09-23:
Correct. I used the full image and force quit the osd and killed the
broadcast program from the shell thinking it would solve my problem. It
didn't. Weeks later I found the solution I mentioned.

rodizio1, 2018-03-31:
I couldn't find out the reason for this, it's strange. As the RT2870 is probably not available anymore anyway, don't really want to add a config option for this as it probably causes more confusion than it helps.

People using the RT2870 and having this issue can simply search for this line: `iw reg set DE` in the /root/.profile script and change it to their needs.

closed 2018-03-31 by rodizio1