rodizio1/EZ-WifiBroadcast #47: error during booting

AAV514, 2017-09-23:
hi, 
I have connected wifi cards on rpi3 and installed driver.
2 wifi cards worked well on latest Raspbian.

then I've installed WFBC 1.5 Image and wifi driver. 
but WFBC doesn't  work.
What is this error means?

[image]

rodizio1, 2017-09-25:
Sorry I can't help you there, Realtek is unsupported and I don't have Realtek hardware to test with.

What I can see from the picture though is, that your wiring and/or power supply is not adequate, it says "throttled=0x50005" (means the Pi firmware detected power problems). See the Wiki for wiring instructions.

AAV514, 2017-09-25:
Yes.  might be power problems. 
I used a  5V 1A, USB charger for test.

closed 2017-10-21 by rodizio1