rodizio1/EZ-WifiBroadcast #49: Huge delay using TL-WN722N V1

ethan2zhong, 2017-09-26:
I got seconds delay using TL-WN722N V1. and another weird thing is that it is getting better when i try to set the frequency more higher,like 2572(maybe 2s delay ) has much lower latency than 2372(maybe 5s delay).the volts are fine and temp is also good. what am i missing here?

rodizio1, 2017-09-26:
This is most likely caused by channels which are already in use. The wifi cards only transmit when the channel is free, if there is not enough  "free airtime" to get the amount of data through, delays are caused.

The 2.4Ghz band is often heavily used, where I live, 2312-2412 is completely unuseable, 2412-2422 is also not really useable because there is somebody streaming video over his wifi accesspoint all day long. 2422-2472 is useable, but I get badblocks caused by other wifi networks around. From 2484-2499 it's good, above that is LTE which also causes delays.

I'd try different channels or go somewhere outside urban areas (or try in an underground parking garage).

ethan2zhong, 2017-09-27:
@rodizio1 Er...I tested every frequency from 2312 to 2572 in the filed,and I couldn't get any improvement. So I dont think it is the "air traffic jam".In the mean time I tested a wifi card with RT3572,and it worked well.

rodizio1, 2017-09-28:
Not sure what it could be then. Since hardware and software that everybody is using is exactly the same and you say wiring and power supply is good, I think it must be some external issue. 

The Ralink cards are less 'sensitive' to external interference (different medium access behaviour). How does it work with the Ralink cards, you should get almost zero packetloss (the 2nd number in the RSSI display) down to around -80dbm, is that the case?

rodizio1, 2017-10-21:
Please try the above things also with 1.6RC1 and open a new issue if the problem persists.

closed 2017-10-21 by rodizio1