rodizio1/EZ-WifiBroadcast #51: bind code for telemetry and RC

geofrancis, 2017-10-07:
can you add a bind function now that you have RC and telemetry, just a unique number or word that can be set on TX and RX so that it wont accept RC or telemetry commands from the wrong groundstation.

rodizio1, 2017-10-10:
For video, telemetry downstream and telemetry upstream, this can be achieved by the wbc 'port' setting (not configurable in 1.5, but would be no big deal making this configurable).

For the R/C uplink it's different, the reason is, that the frames must be as short as possible to keep packetloss low.

I could add another byte for the port to the packets, that would make packetloss slightly worse, which I'm a bit reluctant to do without further testing to quantify it.

Another option would be 'squeezing out' another byte by reducing the resolution of the channels to 10bit=1024 steps (now it's 11bit=2048 steps), but my plan was actually to do that and use the 'freed' byte for additional (low-resolution, for switches) channels.

Now you want to use up that byte for an unique ID feature and all my plans are spoiled ;)

Makes sense though, I'll look into it, maybe I can make it configurable so that people can choose what's more important to them, additional channels, lower packetloss or that unique ID feature.

geofrancis, 2017-10-10:
would it not be easier to just ignore packets not from a specific mac?

rodizio1, 2017-10-10:
That's the problem, there is no mac anymore :)

It's just the 2byte frame control field, two bytes duration field, then the payload (sequence number and channel values). Maybe I can further abuse the 2nd byte of the frame control field, to give maybe 4 different IDs, would that be enough?

geofrancis, 2017-10-10:
should be enough, they can be set with profiles if its in the wifibroadcast.txt

geofrancis, 2017-10-12:
how are the 3dr radios doing it? they have a setting called radio id in the firmware that lets you set 1-100 surely they dont transmit that ever packet?

seeul8er, 2017-10-12:
@geofrancis I have been working on a code that does exactly what you wish. You can set a comm_id so that RC commands on the same frequency do not get mixed up. However there currently is only support for a i6S RC as I do not have access to others.
The code works with Ralink and Atheros cards. The packets have more bytes. I have not tested how much packet loss is caused but the video quality was not affected too much in in my few tests.

You can check out the code here:
[https://github.com/seeul8er/DroneBridge/tree/master/control](url)

bortek, 2018-11-02:
I am trying to clean up the clutter @careyer , do you know if this is still relevant or can we close it ?

careyer, 2018-11-02:
I think we should keep this enhancement since as of now there is no protection for any data that is transfered. Anybody can basically "tune in" to your signal and see your video/telemetry and even can gain control of your aircraft. If packetlength is an issue and must be kept as small as possible - maybe we can simply apply a fast symetric cipher (e.g. AES) to all data that is being transmitted upstream and downstream? High speed and low RAM requirements were criteria of the AES selection process. As the chosen algorithm, AES performed well on a wide variety of hardware, from 8-bit smart cards to high-performance computers. This could at least be offered as a config option. Thoughts?

careyer, 2018-11-02:
@Yes21 : Wolfang writes "... The packets have more bytes (see his post above)." which in my understanding means that the comm_id is added to the payload of each packet as sort of an identifier. Rodizio claimed that this increased packet size would cause considerably more packet loss and hence would not implement this feature. My idea of making use of a fast blockcipher (such as AES) was based on the property that this will not add any bits or bytes (16bit plainext get encoded to 16bit ciphertext) and would make the transmission at the same time immune to "ID enumeration & bruteforcing". There should be cipher implementations readily available for Raspberry Pi.

bortek, 2018-11-03:
Your last suggestion should be good enough security wise if it works practically, we will need to test it out later on. It will ad at least some security to rc communication. Let's keep this for near future.

rodizio1, 2018-11-07:
It's not possible to get that kind of functionality without putting additional information into the packet. Doing the XOR or AES like careyer wrote, would be quite dangerous. Think about it: You encrypt the data and then decrypt it without the ability to check if it was decrypted correctly. When you accidently have a different secret on the RX and TX (or when somebody switches on his WBC R/C system nearby with a different secret configured ...) you end up with garbage being decoded and your controls going completely nuts.

zipray, 2018-11-07:
I used to think that using different FEC values at different frequency points, mainly unique package length values could avoid RC confusion.

bortek, 2018-11-08:
What if you pre-configure the same pre-shared-secret on both ground an airpi. Wouldn't one be able to verify the legitimacy of originating packets?

RespawnDespair, 2018-11-08:
I have always found Vasily's solution pretty elegant. With the generated keys for ground and air.
https://github.com/svpcom/wifibroadcast
I have it on my list to see which parts of his solution could replace some of the parts we have now. I do feel there are some drawbacks, but overall his solution is pretty clean and stable.
We had some talks in the past about two way telemetry, and i feel that can be done, later he also mentioned this himself:
https://github.com/svpcom/wifibroadcast/issues/20

Now i'm not proposing ditching all we have, but his solution merits closer inspection i think...

dposea, 2018-11-13:
Can we take over maybe one packet in 100 for verification? It's not ideally secure but may solve the basic problem. You could also send a block crc for validation. On vasily's site is a link to a white paper on packet injection and you can actually get better throughput with some atheros cards trying to send on the same channel. I'm trying to figure out how we can use that information to improve our uplink abilities