rodizio1/EZ-WifiBroadcast #52: Mavlink Rc with 32 channel taranis

geofrancis, 2017-10-22:
Currently mavlink Rc only works with 8 channels but taranis puts out 32 channels, first 8 are analog and 9-32 are switches that are active when the channel is above 1500.

Now the way mission planner deals with this you can set the switches to do functions and modes like set servo, set relay, set flight mode, arm, takeoff etc

So it frees up the first few channels for other things like gimbal control etc.

rodizio1, 2017-10-26:
I'm not sure on how to implement that as the mavlink_rc_override message the R/C link is using only supports 8 channels. If it's not too much effort I can look into it, but not sure if it will make it into 1.6 final.

careyer, 2018-11-01:
This will be solved with the upcomming new release. Thanks to MAVlink v2 RC_CHHANNEL_OVERRIDE supports now up to 18 Channels. The RC uplink will transport 24Ch in total... 18 of which are fed into the FC and 6 remaining spare to control e.g. GPIOS at the AirPi (hardware actions) or trigger software acitions (e.g. start a script)

closed 2018-11-01 by careyer

Howie9600, 2019-07-30:
was this added in a release?