rodizio1/EZ-WifiBroadcast #54: wifibroadcast data to hotspot computer.

geofrancis, 2017-11-11:
I was out testing a plane with wifibroadcast on it and forgot to take the HDMI monitor with me so i was just using the video screen on mission planner.

I ran across a couple of issues when i was using this set up mainly:

no RSSI or packet information since thats only avalible using the OSD

when I was saving the video after the flight I had no way of knowing when it was finished saving as that is displayed on the osd so i had to just let it sit for 5 minutes to be sure.

the 2 ways i can think of to fix this are,
forward the osd as part of the video stream to the hotspot, that way the osd is always visible on all devices.

support mavlink radio packets and messages, so rssi shows where the 3dr radio data would normally be and send messages like saving video etc as mavlink messages so they flash up on mission planner.

rodizio1, 2017-11-22:
RSSI is currently only supported by the FPV_VR app. Using the the 3DR radio data should be possible to get the info into missionplanner, I'll look into that sometime.

For status messages I'm not sure if it's possible as those messages usually come from the FC, not sure what happens when the Pi would send additional mavlink messages (maybe with different IDs?).

For saving video you could use a usb memory stick with an activity led.

bortek, 2018-11-02:
I am trying to clean up the clutter @careyer , do you know if this is still relevant or can we close it ?

careyer, 2018-11-02:
@bortek : I think the ideas depicted by @geofrancis are well worth to be investigated further. Especially the RSSI portion and the Mavlink Status messages.
As Rodizio pointed out mavlink status messages are usually originated by the FC, so I am not sure if custom/arbitary messages can be implemented easily.
However we should look into a possibility to display Mavlink status/error messages in the OSD, such as "Compass error", "Prearm Check fail: ..." ... This will be of great value when being out on the flying field without a full blown GCS Computer and your Aircraft won't arm for some reason.