rodizio1/EZ-WifiBroadcast #55: record osd with the video

geofrancis, 2017-11-11:
having an option for recording the osd along with the video feed would be very useful for diagnostics and for recovering aircraft.

rodizio1, 2017-11-13:
Unfortunately not possible, that would mean decoding, inserting the OSD and re-encoding the videostream on the Pi.

For recovering aircraft, telemetrylog.txt is saved on a usb stick, it contains all telemetry data like latitude/longitude, height, speed etc.

More fancy would be converting the data into a Google Earth .kml file and saving that to USB, but that's still on my todo list.

closed 2017-11-14 by geofrancis

matiaspl, 2017-11-14:
Not that impossible. There's https://github.com/DougGore/telepi. RPi is
capable of decoding and encoding (and scaling) at the same time.