rodizio1/EZ-WifiBroadcast #56: fast restart of video transmission.

geofrancis, 2017-11-11:
i have been experimenting with a B101 HDMI encoder and a hdmi switcher but i have found that when the video signal is disturbed it wont start again short of rebooting the Tx pi, would it be possible to assign a gpio pin to restart the video capture without loosing RC and telemetry?

rodizio1, 2017-11-13:
I don't have a b101 so I cannot really tell what's happening there. Could be that raspivid hangs or crashes (and needs to be restarted) when the cam is switched. This could probably be detected somehow.

Do you get an errorlog.txt and .png on the TX Pi sdcard? If yes, please post.
If not, please connect a monitor to the TX Pi and make a screenshot after switching HDMI.

matiaspl, 2017-11-14:
I've read somewhere that this is an issue with init procedure of the
Toshiba chip. On newer B101 boards the reset signal is connected to the
GPIO pin that drives the LED on camera boards, so the chip is reset
automatically on raspivid restarts (but the restart has to be triggered
somehow). On older ones you'd have to make the connection yourself (I also
have an older one, so at some point I'll be doing some tests myself)

rodizio1, 2018-02-11:
I'll close this now, feel free to re-open if you have the requested info, can't do anything.

closed 2018-02-11 by rodizio1

geofrancis, 2018-04-19:
sorry its taken so long to get back to you on this I have just rebuilt my machine after a crash. 

i have posted a screen shot of the pi when the hdmi is desturbed or the camera is not on before the pi and B101 HDMI capture.

https://static.rcgroups.net/forums/attachments/3/0/3/8/7/6/a10894708-57-IMG_20180419_024040.jpg

geofrancis, 2018-04-19:
ok from what i have read here: https://auvidea.com/download/manual/B10x_technical_reference_1.4.pdf
the gpio needs to be held low as raspvid is reset.