rodizio1/EZ-WifiBroadcast #57: enhancement: broadcast to multiple devices

roque-canales, 2017-11-12:


seeul8er, 2017-11-12:
Replacing: 

> `ionice -c 1 -n 4 nice -n -10 socat -b $VIDEO_UDP_BLOCKSIZE GOPEN:/root/videofifo2 UDP4-SENDTO:$PHONE_IP:$VIDEO_UDP_PORT &` in [.profile](https://github.com/bortek/EZ-WifiBroadcast/blob/dfe28a2170e51caad57f4fdbb5e8c5fd276fc99b/wifibroadcast-scripts/profile#L1362)

with: 
> `ionice -c 1 -n 4 nice -n -10 socat -b $VIDEO_UDP_BLOCKSIZE GOPEN:/root/videofifo2 UDP4-DATAGRAM:$PHONE_IP:$VIDEO_UDP_PORT,broadcast &`

where` $PHONE_IP` is the broadcast address should do the trick. This would be the easiest way to do it. However if I try it seems that the broadcast transmission can not keep up with the incoming packets. There are a lot of artifacts in the video.
Can someone confirm?

matiaspl, 2017-11-13:
Broadcast and multicast streams don't work well over wifi networks,
tethering might be affected too (not too much info about it on the
internet).
Do have in mind you will be pushing two streams over the same rather slow
connection. Depending on the phone you might have 15-20 mbit of bandwidth
available, so first try to replace $PHONE_IP with the broadcast ip number
and check if you still get errors. However I don't really get the idea why
would you want to do that on a network consisting of WBC RX and a phone...

rodizio1, 2017-11-13:
802.11 doesn't use acks/retransmissions for broadcast and multicast (as it obviously wouldn't make sense), could be that those artifacts are caused by packetloss.

I'm not sure what the cards do with hostapd, but I know that many (if not all? not sure if that is part of the standard) accesspoints send broadcasts/multicasts with a lower (fixed) datarate. If that happens to be too low for the videostream it will probably also delay or glitch.

Total bandwidth (and load on the Pi) wouldn't be a problem with broadcast or multicast, it's the same. With two or more unicast streams it could become a problem though.

seeul8er, 2017-11-14:
Found that the drivers always tend to send broadcast packets with the lowest possible bit rate to make sure every device is able to receive the packets. With 802.11g this is 1Mbps. Looking at the video I can confirm that it seems that the AP simply can not keep up with the incoming data.
Tried to fix it with:

`iw wifihotspot0 set bitrates legacy-2.4 36`
`iw dev wifihotspot0 set mcast_rate 36`
`iwconfig wifihotspot0 rate 11M`

as suggested on [Stackexchange](https://networkengineering.stackexchange.com/questions/1782/why-does-my-udp-broadcast-wireless-communication-is-capped-at-1mbs/).
All return `operation not supported`.

As long someone does patch the driver I do not think we can overcome this issue in the near future. Or maybe I am just missing something :smiley:

rodizio1, 2017-12-28:
Yeah, probably would need patching of the drivers/kernel. I think hostapd also has some options regarding this, but this functionality is of course also dependent on driver/kernel support. Another option would be to use some other cards (which support fixed bitrates in managed or AP mode) for wifihotspot operation.

But that is a can of worms I'm not very willing to open-up right now ;) I'll keep it in mind though ...

closed 2017-12-28 by rodizio1

geofrancis, 2017-12-29:
what about Ethernet hotspot and wifi hotspot together at the same time?

careyer, 2017-12-29:
I don't get what is the point about this feature request? The video (and telemetry) can already be received on as many devices as you want. On all the Groundstations that are just for viewing just disable the Mavlink_Uplink and RC_Uplink and you are fine.

geofrancis, 2017-12-29:
it wont assign an ip to more than one network device last time I tried to connect to ethernet and wifi.