rodizio1/EZ-WifiBroadcast #58: dedicated gpio for specific functions

geofrancis, 2017-11-14:
I have been looking at making profiles for different settings, channels, hotspot data rate etc. 

the problem is the number of profiles multiplies every time you add an option, so say i wanted 4 channels thats 4x profiles, add hotspot on/off and thats now 8 profiles , 2 different data rates, im now at 16.

what i sugggest is having some functions having dedicated pins, so off the top off my head,

3 pins for profiles for resolution/bitrate
3 pins for channels, that gives 8 selectable, you could just have a list in wifibroadcast.txt FREQ1=** FREQ2= etc 
pin for OSD off/on
pin for hotspot on/off

seeul8er, 2017-11-14:
You might be interested in [DroneBridge](https://github.com/seeul8er/DroneBridge). I developed it mainly for private purposes but released it in case others might find it somewhat useful.
DroneBridge extends WBC 1.6 RC2 with a communication module. Together with the [DroneBridge Android App](https://github.com/seeul8er/DroneBridge/wiki/Android-App) you can change settings like the frequency, FECs, resolution or hotspot from an easy to use UI.
The DroneBridge images are fully compatible with the original ones from this repository. 
Beware if you use wbc for telemetry: Some features that involve the relay mode, save to usb or ethernet hotspot of wifibroadcast might not work with the DroneBridge images.

careyer, 2017-11-14:
DroneBridge indeed does look very nice. However besides MSP does it also support MavLink protocol? There are many highly advanced FCs out there that use MavLink as their major control protocol. It would be great to see this supported! ThumpsUp!

geofrancis, 2017-11-15:
yes, i had a look and came to the same conclusion, I use APM and pixhawk flight controllers so msp is no use to me.

seeul8er, 2017-11-15:
Thanks for your interest.
You can change wifibroadcast settings from within the app regardless of your flight controllers software. Same with display of video feed in app. Consider the DroneBridge image as WifiBroadcast + Extras.

MavLink will be supported in the future. MavLink telemetry support of the app and module can be expected till the end of this year. AMP Missions will be introduced together with MSP Missions. This may take me a while.

careyer, 2017-11-16:
@seeul8er, @rodizio1 & @bortek sounds like you guys should team up and integrate code bases to make a huge leap forward. #thumpsUp

seeul8er, 2017-11-16:
If the wish is there I am willing to cooperate.
WifiBroadcast images already act as the base of DroneBridge. So having some common and well defined interfaces might really help. 
With DroneBridge I clearly aim for UAVs relying on MSP (and MAVLink in the future). The android app acts as a user interface.
I am still thinking weather it is a good idea to integrate the DB extensions at this point :thinking:. It might mess up things. Some things might become easier while others get more complicated. E.g. there would be two RC implementations that try to do the same thing but in a very different way.

- If you want simple FPV experience with your goggles --> WifiBroadcast image is enough
- If you want to have something more like the stuff you see with DJIs drones --> (WifiBroadcast Video/OSD + lots of extras) = DroneBridge

If so, at this moment I can only recommend to integrate the DB communication module. This one acts as a interface for the app and does not compete with any other implementations in wbc. That way users can adjust their config fast and easy. I guess all users would profit.
The other modules may change a lot within the next months as they are getting ready for waypoint missions and MAVLink.

Just some thoughts :thinking:

rodizio1, 2017-11-28:
@seeul8er Yeah, a more userfriendly way of configuring things would be good. 

Looking at your protocol and implementation, there are some problems. Json may be nice and convenient for config settings, but it leads to long packets. In upstream direction, packets must be as short as possible or collisions will increase. Your R/C protocol also suffers from this (and doesn't use all cards on the ground pi). Then I don't see the need for all the header info in your protocol, uses up too much bytes. Not a big deal for 1000+ bytes video packets, but for small telemetry and configuration/status data quite a waste. Python is great for easy maintainabilty and readability and quick development, but it uses too much CPU on a Pi1/Pi0 for my taste. May be okay for one-time configuring, but not for repeated transmission of whatever stuff.

Regarding MSP uplink: My new telemetry tx/rx in 1.6 is (almost, still some packet reordering problems) working with Mavlink and Missionplanner, all that is left is to implement parsing for MSP and make changes to the profile script, then EZ-GUI should probably work. If I see it correctly, your code doesnt parse telemetry, and thus has no real control over what chunks of the telemetry messages end up in what wifi frames which is not very optimal.

So yes, we should definitely work together, but please leave the wifi protocol stuff for me ;)

rodizio1, 2017-12-28:
Version 1.6RC3 now supports an additional GPIO pin for a total of 16 config combinations, should hopefully be enough for now. In the long run, configuration will hopefully be made easier.

closed 2017-12-28 by rodizio1