rodizio1/EZ-WifiBroadcast #59: Serial baud rate for OSD

terial, 2017-11-18:
I see that the serial baud rate for the OSD wont go above 57600.... is this something that can be raised in the future? Currently I have my pixhawk connected to the UART at 921600.

rodizio1, 2017-11-18:
Maximum is 115200. Much more won't be possible because of the wifirbroadcast transmission. Maybe 230400 could work. Why do you need such high speeds?

terial, 2017-11-23:
Since i started messing with my RPI and pixhawk I have just always had it set at 921600... I think the ardupilot docs recommended that as well. To be honest I really dont know if I would even notice a difference in data transfer.... from the sounds of it probably not. I have everything setup right now on 115200 and will give it a proper test this weekend.

rodizio1, 2017-12-28:
Did some more testing with telemetry and different baudrates, above 115200 requires special config.txt setting on the Pi and much more than 230400 is not possible anyway.

Considering that 115200 is already twice the speed of the 433MHz Mavlink telemetry radios, I'd rather like to keep it on the safe side and not offer higher speeds than that (could lead to telemetry being delayed if the telemetry rx/tx datarate is not high enough).

closed 2017-12-28 by rodizio1