rodizio1/EZ-WifiBroadcast #60: More dip switch combinations

glatorg, 2017-11-22:
I would like to choose more than 8 different frequencies/configurations on the RX

rodizio1, 2017-11-24:
I'll look into that, another pin for 16 combinations should be no problem.

glatorg, 2017-11-26:
Super! 16 combinations it's ok but 32 should be better because you can cover almost every frequency available for Atheros.

geofrancis, 2017-11-26:
16 is overkill, most channels overlap so you don't need that many

rodizio1, 2017-12-28:
V1.6RC3 now supports 16 combinations. Additional GPIO pin used is BCM1 (or pin 28, see http://pinout.xyz). If you need additional combinations, simply use additonal SD cards as a workaround for now. In the long run, configuration will hopefully be made easier.

closed 2017-12-28 by rodizio1

mathiasseyfert, 2019-03-01:
Is it correct that these pins are ment to be connected to the 3.3V GIPO in order to trigger a profile change? thank you