rodizio1/EZ-WifiBroadcast #63: Disabling the OSD and Telemetry in v1.6RC2

EmbeddedMe, 2017-11-27:
What is the best way to completely disable the OSD and telemetry functions in v1.6RC2?

rodizio1, 2017-11-28:
If you disable all the features in the osdconfig, there should be nothing visible left, although the osd is technically still running then, just not displaying anything. Disabling it completely could be done by putting some syntax error in the osdconfig which would prevent the osd from being started at all. But I've never tested if that may cause other issues, best would be to look through the .profile script and search for 'osd'.

EmbeddedMe, 2017-11-29:
Thank you very much for the tips - I appreciate it. I will look look into your suggestions and see what works best. I just wanted to make sure that I was not missing a more direct method of disabling it.

closed 2017-11-29 by EmbeddedMe