rodizio1/EZ-WifiBroadcast #64: Make IP/Subnet of Hotspot configurable

careyer, 2017-12-01:
Hi there,

unfortunatly the EZ-WifiBroadcast Hotspot configuration is conflicting with my Ethernet settings. Both networks use the same Router IP and the same Subnet resulting in IP address conflicts routing of EZ-WifiBroadcast trafic to my Ethernet router due to lower metrics.

Is there a way to change the Hotspot IP and subnet?
Maybe it is useful to make this configurable -- preferably in apconfig.txt maybe?

Thank you so much & Keep up the terrific work!
Cheers!
careyer

EmbeddedMe, 2017-12-01:
You can change the static IP that is used to configure the hotspot eth0 link in the .profile file located in /root/ on the EZ-WifiBroadcast unit, see the code snippet below where this can be changed.

`	if [ "$ETHERNET_HOTSPOT" == "Y" ]; then
	    # setup hotspot on RPI3 internal ethernet chip
	    nice ifconfig eth0 192.168.XXX.XXX up
	    nice udhcpd -I 192.168.XXX.XXX /etc/udhcpd-eth.conf
	fi`

careyer, 2017-12-04:
@EmbeddedMe Thank you very much... that should solve my problem for now!

rodizio1, 2017-12-10:
There are more places in the .profile script where the network/IP needs to be changed, best is to search through it to find all occurances. 

You also need to change /etc/udhcpd-eth.conf or udhcpd-wlan.conf.

I'll put that on my todo list, but it's not going to make it into 1.6 final.

careyer, 2017-12-11:
Thanks @rodizio1, just yesterday (unfortunately a few hours before your post ;-) we fiddled around with this matter and got it solved exactly like you describe it.

1.) search and replace any 192.168.2.x with 192.168.3.x  (or any other network you wish to use) in /root/.profile
2.) change the addresses in /etc/udhcpd-wlan.conf as well

Works like a charm now! Thank you for supporting this is a future version!
Keep up the terrific work!
Cheers!

careyer

rodizio1, 2017-12-28:
I'll leave this open for later implementation (be warned, could take a while ...)

careyer, 2017-12-29:
Merci! =)  Right now I am using the workaround by editing both files and changing all IPs manually or just pulling the network cable of my home network. For the long term configuring it via the TXT File would come in very handy. Keep up the excellent work!

DaddyK00l, 2018-11-07:
Hello,

While trying to help testing pre beta releases, I confirm having this fixed IP is a real pain... It would be so nice to have a few lines in a config files with just ip subnet getaway dns... Also keyboard in qwerty by default is not friendly for azerty users like me, but this is another problem.

Best regards.

closed 2018-11-23 by careyer