rodizio1/EZ-WifiBroadcast #65: 1mbps bitrate in the receiver side

SepehrKazemian, 2017-12-08:
Hi,

I followed every step that is mentioned in the WifiBroadcast. I just changed the part that needs video on raspberry to record or send with a python code which generates for example only "1"s in a loop and print them all. Then this will be piped to the ./tx with its arguments. The problem is the bitrate in the receiver side which is less than 1mbps (as I saw in its radiotap headers) based on wireshark sniffings. Just in case, I am using two raspberry PI3 and two atheros dongles (the same version that is suggested).

Thanks in advance

rodizio1, 2017-12-10:
Less than 1Mbit is not possible as the lowest bitrate is 1Mbit. If the cards send with 1Mbit wifi datarate, you probably have not patched the firmware. Either use my images (they include a properly patched Atheros firmware) or just take the firmware from my image. Or compile it yourself as described on Befinitiv's blog.

If you want to measure the actual throughput you get, you can for example do:
"cat /dev/zero | pv -i 0.5 -r | ./tx ...." or "your_python_script | pv -i 0.5 -r | ./tx ...."

SepehrKazemian, 2017-12-10:
Hi,
Thanks for response,
First of all, Wireshark shows 1mbps and TCPDump shows less than that. The point is less than 1mbps is not impossible since it is a basic data rate and if the sender sends data less than that it would be possible. 
After that, I double checked and the images didn't work but I did the compilation and it worked well. One problem stays on. Reciever gets the data in an interval of "data rate that was specified in the radiotap headers" and "1mbps". For example, the receiver gets data in 1 and 17, again 1 and 17, ...
It works oddly.

rodizio1, 2017-12-28:
Yes of course, if you send only let's say 10 packets per second, actual datarate is lower than 1Mbit. What I meant was the frame datarate that is shown in the radiotap header.

If you get "alternating" frames with my Atheros firmware, that's because it generates CTS-to-self protection frames when you send ieee802.11 data frames (which is what Befinitiv's original wifibroadcast does). If you want to send traffic without CTS, you can send RTS frames like my tx_rawsock wifibroadcast implementation does.

In general, if you want to make changes, I'd recommend using my image and the sources contained in there as a base. This way, kernel, firmware and wifibroadcast software fits together, has been extensively tested and you get all the improvements that have been made over the years.

closed 2017-12-28 by rodizio1