rodizio1/EZ-WifiBroadcast #67: How to hide "Ethernet connected"

glatorg, 2017-12-20:
I would like to keep the ethernet connected but hiding the message "Ethernet connected IP...." .
Someone can point me in the right direction?

seeul8er, 2017-12-22:
`nice /root/wifibroadcast_status/wbc_status "Ethernet connected. IP: $ETHCLIENTIP" 7 55 0`
[link to file on github](https://github.com/bortek/EZ-WifiBroadcast/blob/dfe28a2170e51caad57f4fdbb5e8c5fd276fc99b/wifibroadcast-scripts/profile#L1887)
This file is located in `/root` and called `.profile`. 
Just edit/delete the line

rodizio1, 2017-12-28:
You also need to comment the lines that re-start the osd process before and after the wbc_status ... line if you don't want to have the osd disappear for a short time.

closed 2017-12-28 by rodizio1