rodizio1/EZ-WifiBroadcast #69: Streaming data from IP cam

tbeauchant, 2017-12-21:
Hi!

First of all, thanks for your great work on this project! It has really reached an impressive level of quality!

I am trying to get an IP cam that provides rtsp streams of H264 video to work along wifibroadcast.
At this stage, I have a working wbc system with two RPi3s and AWUS051NHV2s.
I have connected the IP cam to the TX Pi through ethernet, setup IPs and all, and can read correctly the stream with gstreamer.
The Pi being headless I verified this by storing the video in an mp4 file and checking the file:
gst-launch-1.0 -e rtspsrc location=... ! rtph264depay ! h264parse ! mp4mux ! filesink location=...

I then edited the .profile file to pipe the output h264 stream from gstreamer into tx_rawsock:
gst-launch-1.0 -e rtspsrc location=... ! rtph264depay ! filesink location=/dev/stdsout | tx_rawsock ....

I can see the data being sent out and received on the receiver end (approx 3.5MB/secs of incoming data) with variations in datarate when there is movement in front of the camera (so it does appear like a real stream of data). However I am unable to get gstreamer on the receiver side to display any picture.

Where this gets strange is that when I replace the rtsp stream in gstreamer with a videotestsrc, encode it in h264 and pipe it to tx_rawsock, I do get the stream displayed on the receiver.
gst-launch-1.0 videotestsrc ! omxh264enc ! filesink location=/dev/stdout | tx_rawsock ...

I have tested transmitting directly the rtp stream through wbc (instead of transmitting only the h264 data) and have got plenty of errors on the receiver side complaining about corrupted rtp frames.

I have also tested on a low resolution stream from the camera (approx 850KB/secs) and got the same result, so I wouldn't think datarate is an issue here.

Is there any reason to this that would come to mind? Any pointers on where to look would be greatly appreciated!!

Thanks,
Theo

rodizio1, 2017-12-28:
Sorry I have no experience with IP cams nor much knowledge about gstreamer. I'd try narrowing it down first by leaving wbc out of the equation by piping the gst-launch output videostream directly (or via fifos) in the videoplayer and see if that works. Maybe also use the "pipeviewer" tool to see what's going on, you can put it in-between a commandline like "gst-launch .... | pv -i 0.5 -r | ./tx_rawsock ...." (or similarly on the rx side) to see if data is coming in and at what bitrate.

tbeauchant, 2017-12-31:
Hi Rodizio,

Thanks a lot for your input, that is much appreciated!

I've kept making tests and I am now quite sure wbc is not the culprit, it seems to be a gstreamer issue. 

I do have another question for you however: When I put the transmitter on my spectrum analyzer, I only see about 20dBm of output power. I have tested using both an AWUS036NHA and 051 cards with the same result. I believed from docs that on default config the output power was set to maximum which should be higher than 100mW using those cards. Any idea on how to set it higher?

Many thanks, and happy new year!
Theo

rodizio1, 2018-01-01:
Hmm, maybe try using a newer gstreamer version. Would require compiling from source though, could be quite some work. Or maybe try on another computer with another Linux and gstreamer version before you put in all the work on the Pi.

Regarding the transmit power. Somebody on the RCGroups forum once measured about 24.5dbm, that was with the txpower parameter set to "60" and default wifi bitrate 18mbit. 1.6RC3 comes with the parameter set to 58 (1db less in theory), so it should still be above 200mW.

Is your wiring and powersupply good?Wifi bitrate also affects output power, did you measure with stock wifi bitrate settings? You could try setting 11mbit mode, that should give higher values. Also set CTS protection to disabled when measuring. Does your spectrum analyzer have some kind of hold mode that is quick enough to "see" the packets which are less than 1ms duration?

Maximum transmit power settable is "63", see the wiki on how to change the settings.

tictag, 2018-01-02:
Hey Thio,
Just wanted to say that I'm following this thread with great interest. I am trying to do something at least a bit similar ... trying to get a stream output from OpenCV (which in turn is a thermal video stream from a FLIR Lepton) to be broadcast via wifibroadcast. I'm a total novice at all this but I'm going to follow the approach you outline at the top using gstreamer to 'integrate' OpenCV and wifibroadcast. It's all new and exciting. Wish me luck! :)

careyer, 2018-01-02:
@tictag and @tbeauchant sounds very promising what you are trying to achieve. We want to try something similar. However we prefer to go with a FlirOne (Thermal camera for Android). You might want to check out: http://diydrones.com/profiles/blogs/thermal-imaging-for-your-drone-on-a-budget

The FlirOne is also based on a Lepton sensor but features a better thermal resolution and also a HD (visual light) overlay camera. There is already a Raspberry/Linux-Driver for this setup available. It would be great if we could exchange experiences and progress when it comes to piping these alternative video signals to EZ-WifiBroadcast. Cheers!

tictag, 2018-01-02:
@careyer Very pleased to meet you! Interesting, I'm actually trying to exactly the same thing! But I don't want to hijack Thio's thread discussing it. If interested in my drone thermal imaging solution, head over to https://mavicpilots.com/threads/thermal-imaging-on-the-mavic-pro.17116/

bortek, 2018-01-02:
Indeed this is interesting. I dont own any FLIR camera but have SeekThermal for Android. Some time ago I looked if someone integrated it with Raspberry Pi but could not find any projects. Perhaps any of you had heard more about it ? :)

careyer, 2018-01-03:
Maybe it makes sense to start a dedicated thermal imaging thread here - not in terms of an "issue" but to exchange experiences and progress of integration with EZ-Eifibroadcast? This could be useful.
@bortek : The only indication that the Seek Line of cameras works with Raspberry I found here: http://diydrones.com/forum/topics/thermal-camera-works-in-mission-planner-hud - However this guy seems a bit strange to me. I got in touch with him and he somehow made a business out of this achievement and wants to see test-units.

tbeauchant, 2018-01-03:
Hey Folks,

@rodizio1 : Thanks for the tips and for the info on the transmit power, I'll test it out as soon as possible and get back to you with measurements.

@tictag @careyer @bortek : I don't have any experience with FLIR USB cameras, but I'd suggest testing on a linux PC first and trying to get a video stream from the camera. Once you have it try to move it to a raspberry pi, you will probably need to recompile on the Pi any driver/software that was required to get the camera working. Once you have an accessible video stream from the camera you can use gstreamer to shape it in whatever format you like and forward this to wbc's transmit software.
One thing I love about wbc is that it is pretty much data agnostic, you can send whatever data you like thorugh it and get it on the rx side.

I don't have any camera to do some testing, but don't hesitate to PM me with questions if you get stuck trying to make this working.

Cheers,
Theo

careyer, 2018-01-22:
@tbeauchant : sorry I do not know how to PM you. Since you succeeded in getting a gstreamer pipeline running via EZ-WifiBroadcast: Maybe you can help me over there #71 ? Thanks a lot!

tbeauchant, 2018-03-01:
Hi All,

Just wanted to give a little heads up on all these matters and close the topic.

It looks like the RTP protocol doesn't like to be pushed into a continuous stream of data and one has to restore the UDP packets with their original length for gstreamer to be able to decode it. I don't know if this is a limitation of the RTP protocol itself or gstreamer's implementation but modifying tx_rawsock to maintain the original packet's length and output UDP packets just the way they were seems to do the trick. This does seem to wreak havoc with bandwidth though (at least with my quick'n dirty mod).

@rodizio1 Also, on the output power side, I have just tested the ubiquiti wifistation card and I am getting 27dB of output power. My spectrum analyzer wasn't calibrated in a while and I had a few SMA/RP-SMA/N adapters in the loop that would certainly account for the missing few dBs.

Anyway, thanks to all of you for your help! And again, thanks a bunch for providing WBC to the mass, this is truly amazing work.

Thanks,
Theo

closed 2018-03-01 by tbeauchant

nsimon1, 2018-03-26:
Hello @tbeauchant,

I am having the same issue where I need to stream the H264 video from a RTSP source using tx_rawsock. Would you mind elaborating on how you modified tx_rawsock to keep the same packet lengths? If you have the source uploaded somewhere I would appreciate having a look at it and trying it out. 

Thanks!

tbeauchant, 2018-03-28:
Hi @nsimon1 ,

Unfortunately I lost the source code a few days ago when I accidentally rolled my chair over the micro sd card of the Pi (how stupid can it get......).
The way I modified tx_rawsock for this was initially very crude. I added a UDP listener socket, and each packet received was placed into one of the video blocks to be transmitted.
On the receiving end, every video block received was sent through a single UDP packet.

The modifications to achieve this behaviour are pretty simple really. However, doing so seems to wreak havoc with bandwidth (say you receive a very small UDP packet, then you loose an entire video block transmitting it), so a more thorough work would be needed to do this properly.

At the moment, I am sending the h264 stream through an mpegts encapsulation. This works fine with the way wbc works out of the box, but does add some latency (~400ms I'd say). Not an issue in my application but it might be for you.

Cheers,
Theo

nsimon1, 2018-03-28:
@tbeauchant I am sorry to hear about the SD card, it is unfortunate! Thank you for the additional details, I will have a look at your suggestion of using mpegts encapsulation as well, however the latency would be an issue for me if I cannot get it decreased. 

Thank you for the help!