rodizio1/EZ-WifiBroadcast #71: Using OpenCV with EZ-WifiBroadcast

tictag, 2017-12-28:
Hello Rodizio,

Just got my first video stream to work [_RPi Zero()W with ZeroCam to RPi3 using 2 x RT5370 based nano USB adapters_] and it looks great! :) Really fantastic work (from you and Befinitiv)!!

My query **does not** need a comprehensive answer, I don't mind doing all the digging to get something working, but I don't want to waste my time investigating if my use case simply isn't an option. I plan to take two seperate video sources connected to a RPi, do some rudimentary image processing then merge them together using OpenCV  ...  and here's the query  ...  could I then use EZ-WifiBroadcast to transmit that composite video stream to a receiver?

I've read every page of your wiki and everything revolves around broadcasting a source video from Tx to Rx. Basically, can my source video actually be an output stream from OpenCV?

I don't mind putting the hours in to investigate/troubleshoot etc but not if this use case is simply not do'able.

I would appreciate your thoughts.

Oh, and if you don't get back to me before new year, Happy New year!! :)

rodizio1, 2017-12-31:
The wifibroadcast tx (as well as my further developed tx_rawsock) basically just send out what they receive on stdin, so if you can pipe your OpenCV output into it somehow it should work.

tictag, 2018-01-02:
Ahhh, so it isn't just 'hard wired' to transmit the camera stream, anything 'piped' (_I'm not 100% sure what that means just yet_) into tx_rawsock should be transmitted. That sounds brilliant. I've just read tbeauchant's post re an IP cam and s/he is using gstreamer ...

> I then edited the .profile file to pipe the output h264 stream from gstreamer into tx_rawsock

Looks like I have my starting point :)   Do you have any further docs around tx_rawsock?

Many thanks for all your (and Befinitiv's) hard work on this. Seriously, there is no way I'd ever be able to complete this project without you.

ivanskj, 2018-01-02:
I am piping a gstreamer-pipeline from a Sony QX1 MJPEG liveview-url to tx. Souphttpsrc -> h264enc. I can confirm that it works, but i still have to figure out bitrates and all that stuff to get more range. There is also some more latency on the stream. I will try the tx_rawsock

geofrancis, 2018-01-02:
i would be very interested in using a sony camera with wifibroadcast, if you could switch between it and the pi camera it would be awsome.

tictag, 2018-01-04:
rodizio1,

Whilst I understand the purpose of EZ-WiFiBroadcast is simplify setup i.e. just download a couple of SD card images and you're good to go, I'm going to need to install a whole lot of software to complete my project e.g. OpenCV, Numpy, Python etc. Is there a 'manual install' doc available? That is, a doc that outlines the steps required to essentially create a working image from, for example, a vanilla Raspbian distribution like Raspbian Stretch Lite?

careyer, 2018-01-04:
Maybe the easier way to go is to find a way to boot the EZ-WifiBroadcast image so that it does not autostart all the RX, TX, OSD and Telemetry processes but welcomes you with a login prompt. Connecting the ethernet port to your router would allow for installing all the necessary packages then.

@rodizio1: is there a best practice how to boot the image so that it does not enter transmission / reception mode automatically?

tictag, 2018-01-08:
The second method simply provides you with a remote command prompt much in the same way as CTRL-C does locally.

It would certainly be better for me to have the manual install instructions. I've been trying to install and configure stuff on top of the image but just keep running into problems; bash commands that are not present, make commands (e.g. numpy) that appear to run/compile forever (48 hours before quit, should be 3-4 hours). And of course I'm scared to do any kind of apt-get update/upgrade for fear of modified drivers/firmware being overwritten.

Whilst I do certainly believe that some would prefer the EZ (image-based) installation, for others this might cause more problems than it solves.

It would be great to have a manual install method.

rodizio1, 2018-01-08:
There are no instructions as I just made changes to the image and did not document them apart from the changelog. Instructions on howto make the changes would need to be constantly updated as Raspbian is constantly changing, I don't have the time nor motivation for that. Finding out what has changed is quite easy, simply take the default raspbian lite image and compare it with my image. "meld" is a good tool for that for example.

In the long run, I want to get away from Raspbian and use something smaller and better manageable like buildroot.

In general:
- Do not run apt-get upgrade, it'll probably just overwrite a lot of stuff and break everything
- Do not upgrade the kernel, chances are you run into whatever issues
- If you want to make changes to the kernel, build the one that is included in the image (instructions and patches/config in the /kernel/ directory on the Github repository)
- Do not change Raspberry firmware or Atheros firmware
- Installing software packages using apt-get is possible. Just make sure no existing packages are upgraded (it tells you before). If you cannot get around upgrading packages, check before which ones are affected so that you can act accordingly (like making backups of config files that may get overwritten). Also make sure you don't accidently install whatever stuff that messes with network interfaces like network-manager etc. or stuff that installs daemons that may cause whatever issue when they do stuff in the background

Regarding logging in: Hmm, the OSD still being in front of the console when logging in locally is indeed not so nice. I've changed the profile script so that it automatically does a "killall osd" to quit the OSD after logging in.

tictag, 2018-01-09:
Wow, so this has been a voyage of discovery!! I have now managed to get python3.4.2, numpy1.8.2, gstreamer1.0 and opencv3.4.0 all installed and working simultaneously with EZ-WiFiBroadcast v1.6RC3 on a Raspberry Pi 3. This has required me to: resize file systems, find random files, compile code(!), edit configuration files and solve many other problems along the way but ... at least it works!

I'm working on a fully documented bash script to automate this whole thing and I'll upload once I've tested it for others to use should they wish to.

rodizio1 thank you for your candour, I totally get it. As I am eventually going to be writing an instructable for my project, I will want to start off with a vanilla Raspbian image, that is, I am going to try to use that "meld" tool. What was the exact version of original Raspbian source image?

tictag, 2018-01-09:
_...p.s. I only put the (!) after 'compile code' because **this is the first time I have ever compiled code**. Yep, that's how much of a layman I am!_

Now that I have everything I need, I'm gonna have to stop wifibroadcast from snagging the hardware (i.e. camera) and instead have OpenCV do this. OpenCV then needs to process the video streams before piping them out through gstreamer to the wifibroadcast tx_rawsock device.

Why does it sound so easy when it's just words? ;)

lgangitano, 2018-01-11:
@tictag I'm interested in your solution, since I'm experimenting with the same setup (video feed multicasted to OpenCV and EZ-Wifibroadcast) for realtime object tracking and streaming. Would you share your results performance-wise on Pi3?

tictag, 2018-01-12:
Of course, happy to. I'm just at the point where I'm adding my thermal stream so will be, probably tomorrow be looking to piping the streams into wifibroadcast, as opposed to it capturing the stream itself. On the receive side, I'll be extracting the two streams from wifibroadcast and piping them into OpenCV for further processing.

...and I have no idea how to do this yet!! Don't let me sound like I know what I'm doing! ;)

tictag, 2018-01-15:
> rodizio1 thank you for your candour, I totally get it. As I am eventually going to be writing an instructable for my project, I will want to start off with a vanilla Raspbian image, that is, I am going to try to use that "meld" tool. **What was the exact version of original Raspbian source image?**

Bump...

rodizio1, 2018-01-15:
Sorry, I never wrote that down (and in retrospect I found out that there is no version number or similar inside the Raspbian images ...)

What I remember is, that version 1.0 was released around 15th of May 2016 and used Kernel 4.4.9 or 4.4.11, so it must be a Raspbian release around that time with that kernel.

careyer, 2018-01-16:
@tictag : I am very interested in the automated bash script that you created to automate the installation of additional components to EZ-WifiBroadcast1.6RC3

> I'm working on a fully documented bash script to automate this whole thing and I'll upload once I've tested it for others to use should they wish to.

In my use case I need to install the following components:
- GStreamer
- v2l2loopback driver (https://github.com/umlaeute/v4l2loopback) - This needs to recompile the kernel, have no idea yet how this works.
- FlirOne Device Driver (http://www.eevblog.com/forum/thermal-imaging/question-about-flir-one-for-android/?action=dlattach;attach=272327)

Being a total Linux noob this might help me get started a bit less troublesome. Thank you very much!

BTW: Patrick Duffy from DIY Drones (http://diydrones.com/profiles/blogs/thermal-imaging-for-your-drone-on-a-budget) send me ready-2-run image demoing the integration of FlirOne with Raspberry. It works flawlessly. However it does not build on WiFiBroadcast but on nomal WiFi streaming via Gstreamer and fixed IPs over the local WLAN.

tictag, 2018-01-16:
Happy to help a fellow noob! If you only need gstreamer then my script probably won't help so much. Tbh, the most complicated thing (for me) has been compiling OpenCV. Mind you, should help with compiling the kernel. Definitely we can work together on this :) (... blind leading the blind ;)

careyer, 2018-01-16:
@tictag : That is good news! I will try to install gstreamer first (I suppose the way to go is to do it the regular way with apt-get?) and then get back to you? I think the FlirOne Device driver needs to be compiled as well :-|.  Am I correct that you are using this driver for your project as well? Last (and probably most complicated) will be recompiling the kernel in order to get l4l2loopback support added. For this I definitely need some help.

BTW: Here is a screenshot from what I was able to achieve yesterday evening with the Patrick Duffy image. The frame rate was surprisingly good - I believe it was definitely more than the regular 8-9fps. Felt more like 15-20fps which I was positively surprised about: [image]

tictag, 2018-01-16:
Looking hot! Ahem, sorry...

Yes, gstreamer just via apt-get, though I did have a few issues installing:
- Missing packages during installs
- Running out of diskspace installing most things
- 'Fuse' directory not found during gstreamer install

...to resolve
```
# Missing packages during installs
apt-get update  # do not 'upgrade'

# Running out of diskspace installing most things
nano /etc/fstab  # for device /tmp (ramdisk used as temp scratchdisk), change size=50M, CTRL-X, Y to save then reboot
resize2fs /dev/mmcblk0p2  # this is the 2nd sd card partition and mounted at /dev/root, which is your home directory
df -f  # make sure /dev/root size is the same size as your partition and there's plenty space 'avail'

# 'Fuse' directory not found during gstreamer install
mkdir /dev/fuse
chmod 777 /dev/fuse
apt-get install fuse
apt-get install gstreamer1.0
```
That should get you gstreamer installed.

tictag, 2018-01-16:
...oh, forgot to answer your question, no I'm not using the FLIROne, I'm just using the [FLIR Lepton OEM module](http://www.flir.com/cores/lepton/) on a [breakout board](https://groupgets.com/manufacturers/getlab/products/flir-lepton-breakout-board-v1-4), connected to a RPi Zero via SPI (_Serial Peripheral Interface_), I'm using [pylepton](https://github.com/groupgets/pylepton) to control it, OpenCV for image processing and gstreamer to stream it via wifibroadcast's tx_rawsock.

Does the FLIROne connect via USB?

careyer, 2018-01-16:
Yes, the FlirOne conencts via USB and features both a Thermal camera and a HD cam for visable light. Both videostreams can be accessed with the linux driver.

rodizio1, 2018-01-16:
say, what are you guys installing there exactly? Gstreamer is already installed (used for encapsulating the raw h264 video for missionplanner) and there is no package called "gstreamer1.0"

careyer, 2018-01-16:
Hi rodizio... We are trying to get some FLIR Thermo cameras working with EZ-Wifibroadcast. In order to make the Flir cameras work we need a very complicated Gstreamer pipeline with modules from Gstreamer good, bad and ugly... that is why we need the fullblown Gstreamer suite. apt-get install gstreamer1.0 installes the hole suite and works flawlessley as far as I can tell. It installs all the nessesary additional modules. 

I also succeded to install the FlirOne USB driver (flir8p1-gpl), however now I am stuck installing the v4l2loopback kernel module. :-(

I did:
```
sudo apt-get install linux-headers-rpi
 sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update
 rpi-source
 sudo apt-get install bc
 rpi-source

git clone https://github.com/umlaeute/v4l2loopback
 cd v4l2loopback
 make
 make install
```

Everything runs just fine without any errors but the kernelmodule v4l2loopback wont show up:

```
**root@wifibroadcast(rw):~/v4l2loopback#** sudo modprobe v4l2loopback
modprobe: FATAL: Module v4l2loopback not found.
**root@wifibroadcast(rw):~/v4l2loopback#**
```

any ideas?

careyer, 2018-01-17:
@rodizio1 :  I think I know now what the problem is with compiling the v4l2loopback kernel module

The commands above load the kernel sources for 4.9.35-v7+ wheras EZ-Wifibroastcast v1.6RC3 runs on 4.9.35-v7 (non+). That is why the kernel module does not get installed properly.

Can you please advice me how to install the correct kernel sources (4.9.35-v7) ((and how exactly to apply the patches you made). I took a look at https://github.com/bortek/EZ-WifiBroadcast/tree/master/kernel/readme.md but I do not completly understand how to do it. You know... I am just a linux noob! THX!!

careyer, 2018-01-17:
Okay success!
I finally solved the problem...
_rpi-source_ always adds a '+' to the release number of downloaded kernels from Github to indicate that it is a local copy. The make process of the _v4l2loopback_ kernel module copes just fine with that but unfortuantly _modprobe_ gets into serious trouble if the kernel release string differs only slightly. It simply won't start and _make install_ will also copy it to a wrong destination.... GRRRR!! I had to manually alter the release  number string all over the filesystem (hidden files, config files.... god knows were I found this damn attached '+'). Now it works!
FlirOne is now functional in the EZ-WifiBroadcast Image 1.6RC3... Now I just have to figure out how to pipe its output to the tx pipeline.

Cheers!

careyer, 2018-01-21:
@rodizio1 / @ivanskj / @tictag 
I was finally able to get all the prerequisites for operating the FlirOne thermal camera right. Driver is installed and is linked to the video3 devices on the system. I also have a working gstreamer pipeline. However as of now this gets streamed via a standard WiFi connection (via a seperate NIC) and not via WifiBroadcast.

I do not fully understand yet, how I can pipe my Gstreamer Output to WiFi-Broadcast and have it transfered. Can you please help me to make that happen and guide me a little bit on what needs to be changed and where? That would be awesome!

My Gstreamer pipeline looks like this now:
`gst-launch-1.0 v4l2src device=/dev/video3 ! video/x-raw,width=160,height=128, framerate=1/10 ! rtpvrawpay ! udpsink host=192.168.0.100 port=9002`

Cheers!

tictag, 2018-01-21:
I've yet to figure this out myself but I do have a novice understanding about 'piping' in Linux. Piping is where you take the output of one process and feed it to the input of another, ever see that cult film 'the human centipede'? So, 

`ls -p | directory.txt`

... would output the current directory list and 'pipe' it into a file instead of printing it to the console.

I've seen examples on the t'interweb where the output of raspivid (the default application used for pi cameras) is then 'piped' into the tx_rawsock application, thus:

`raspivid <parameters> | tx_rawsock <parameters>`

I'm not quite sure yet how to do this with OpenCV but as I'll be doing the image processing using Python, I'll likely be using the `cv.ImageWriter()` method, piped into gstreamer, which in turn pipes into tx_rawsock.

Btw, if you have seen 'the human centipede', I apologise for associatibg that imagery with Linux piping!! ;)

ivanskj, 2018-01-21:
Use fdsink as the last element in gstreamer

Sendt fra min iPhone

tbeauchant, 2018-01-22:
@tictag @careyer 

Piping is the process of grabbing the stdout of one program and feeding it into the stdin of another. In the .profile script this is what is achieved on lines 688 & 71:
raspivid ... | tx_rawsock

Raspivid outputs to it's stdout the h264 data, this data is grabbed and sent to the stdin of tx_rawsock which in turns does it's magic to send it through the wifi.

if you'd like to achieve the same thing with gstreamer, you'll have to replace the raspivid part of the command with your gstreamer pipeline, and make sure gstreamer outputs to stdout. This can be achieved by using the following sink:
gstreamer .... ! filesink location=/dev/stdout
Be aware that the video stream is not the only data that gstreamer will output on stdout. Every line displayed in the console when you run the software is also on stdout, so depending on situations, this may or may not cause issues in the datastream.

In order to get rid of this issue, I have created a fifo using mkfifo, and modified tx_rawsock to use it (look in the source code, there is a line where it opens the /dev/stdin, replace it with the name of the fifo file you created). Then in gstreamer simply output the data to the fifo: filesink location=/path/to/my/fifo

A fifo is simply a special type of file on the filesystem that acts as a buffer.

Also, the reason tx_rawsock is used twice in the .profile is to output different texts messages on the display dpeending on whether the script has detected some issues or not. Usually for testing I disable both these lines and run the command line myself from the terminal.

Good luck,
Theo

careyer, 2018-01-22:
@tbeauchant : Great I did some research on my own and came up with a similar approach in the meantime. However your hint with the fifo-buffer is worth gold 🥇. Sounds like the implementation on on the AirPi side of things is straight forward and quite easy. 

What worries me most is how to process the gstreamer data at the GroundPi. I do not understand what this line of code exactly does and where the actual video decoding and displaying via HDMI takes place.
>Line 837 in .profile
ionice -c 1 -n 3 /root/wifibroadcast/rx -p 0 -d 1 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH $NICS | ionice -c 1 -n 4 nice -n -10 tee >(ionice -c 1 -n 4 nice -n -10 /root/wifibroadcast_misc/ftee /root/videofifo2 > /dev/null 2>&1) >(ionice -c 1 nice -n -10 /root/wifibroadcast_misc/ftee /root/videofifo4 > /dev/null 2>&1) >(ionice -c 3 nice /root/wifibroadcast_misc/ftee /root/videofifo3 > /dev/null 2>&1) | ionice -c 1 -n 4 nice -n -10 /root/wifibroadcast_misc/ftee /root/videofifo1 > /dev/null 2>&1

I guess I have to put my gstreamer decoding pipleline somewhere .. but where?
Thanks in advance! - Thomas

rodizio1, 2018-01-23:
I have explained that tee/ftee/fifo thing sometime ago on the rcgroups thread (for telemetry but it's the same principle for video), if you search for my username and 'ftee' or 'telemetryfifo', you should find it.

For using gstreamer instead of hello_video for display, check the 'DISPLAY_PROGRAM' variable, it should be possible to change that (it's set depending on fps configured, you probably don't need that with gstreamer).

Regarding the kernel: I think the easier and cleaner way would've been not using rpi-source and simply downloading and extracting/compiling the sources from the link in the /kernel directory readme here.

careyer, 2018-01-23:
> @rodizio1 : I have explained that tee/ftee/fifo thing sometime ago on the rcgroups thread (for telemetry but it's the same principle for video), if you search for my username and 'ftee' or 'telemetryfifo', you should find it.

Thank you very much! I think I found it. For better reference I will link it here: https://www.rcgroups.com/forums/showpost.php?p=36551372&postcount=1897 . As far as I understand now I do not need to change anything at the tee-mimic. Just need to replace the displaying program?

> @rodizio1 : For using gstreamer instead of hello_video for display, check the 'DISPLAY_PROGRAM' variable, it should be possible to change that (it's set depending on fps configured, you probably don't need that with gstreamer).

Hilarious, also found that in .profile Line 1644-1658:
```
if [ "$FPS" == "59.9" ]; then
    DISPLAY_PROGRAM=/opt/vc/src/hello_pi/hello_video/hello_video.bin.48-mm
else

    if [ "$FPS" -eq 30 ]; then
	DISPLAY_PROGRAM=/opt/vc/src/hello_pi/hello_video/hello_video.bin.30-mm
    fi
    if [ "$FPS" -lt 60 ]; then
	DISPLAY_PROGRAM=/opt/vc/src/hello_pi/hello_video/hello_video.bin.48-mm
#	DISPLAY_PROGRAM=/opt/vc/src/hello_pi/hello_video/hello_video.bin.240-befi
    fi
    if [ "$FPS" -gt 60 ]; then
	DISPLAY_PROGRAM=/opt/vc/src/hello_pi/hello_video/hello_video.bin.240-befi
    fi
fi
```
I can replace hello_video.bin with the gstreamer here. I am just wondering how I can specify all of the gstreamer pipeline parameters here? 
(they are quite massive: `udpsrc port=9002 caps="application/x-rtp,media=(string)video,clock-rate=(int)90000,encoding-name=(string)RAW, sampling=(string) YCbCr-4:2:0, depth=(string)8, width=(string)160, height=(string)128, colorimetry=(string)BT601-5, payload=(int) 96" ! rtpvrawdepay ! autovideosink`)

> @rodizio1 : Regarding the kernel: I think the easier and cleaner way would've been not using rpi-source and simply downloading and extracting/compiling the sources from the link in the /kernel directory readme here.

Thankfully this problem is solved. In fact I did something similar to your suggestion. I used rpi-source with the "--uri" parameter and specified the  Raspberry 4.9.35 kernel sources you have linked in this very Git repo under /kernel.  Works just fine. Only problem was that rpi-source adds a "+" to the kernel.release String to indicate that this was a kernel downloaded from a GitHub source. In consequence of this the kernel module I needed to compile was compiled with a differing kernel.release string causing problems at runtime. Problem could be solved by eliminating the "+" in three different files manually before the compile. Works like a charm ;-).

rodizio1, 2018-01-28:
> Thank you very much! I think I found it. For better reference I will link it here: https://www.rcgroups.com/forums/showpost.php?p=36551372&postcount=1897 . 

I meant this one, it describes the telemetry part, but it's the same logic for video:
https://www.rcgroups.com/forums/showpost.php?p=37651300&postcount=2792

> As far as I understand now I do not need to change anything at the tee-mimic. Just need to replace the displaying program?

It depends on what you want to do and how you want to do that. Replacing the display program with your gstreamer should work. But then you only have one stream at the rx? I thought you guys wanted to send two different videostreams to the rx? In that case you'd have to run another rx and the tee/ftee/fifo logic described in the post above.

I'd recommend reading some bash scripting tutorial and maybe also a basic tutorial that explains some basic unix concepts like I/O, redirection, pipes, stdin/stdout/stderr, shell, processes, some basic shell commands, special characters and escaping them etc.

careyer, 2018-01-28:
@rodizio1 : Thank you very much! I followed your recommendation and studied all sorts of things (tutorials, manpages...) that I could find with a little help from my friend Google! =) Indeed I made some considerable progress, which is: I could successfully transfer the gstreamer datastream via tx_rawsock to the ground. The OSD on my GroundPi shows that those packets indeed do arrive. Since the FlirOne has only a low resolution of 160x120 Pixels it needs considerably less bandwidth which is also reflected in a lower packet count displayed via the OSD. So far so good 👍 

The very last thing im am struggling with is to display the received data via HDMI. Somehow I do not really understand yet where the data is being displayed. I think i try to do it at the wrong location in the code.

May I kindly ask you to have a quick glance at my edited .profile file? I introduced a new Variable
`FLIRONE="Y"` at the top of the file and marked each location that I have edited with `### FLIRONE ###` to allow for any easy understanding of what has been changed.
All edits edits/changes follow the concept:
```
    ### FLIRONE ###
    if [ "$FLIRONE" == "Y" ]; then
       new code 
    else 
       original code
    fi
```
Here is a link to my File: http://www.ip-networx.de/content/GitHub/editedProfile.txt

Your help is so much appreciated! Thanks in advance!
Cheers!
Thomas

ivanskj, 2018-01-28:
I am also transmitting a lower res picture via wbc. It is the liveview from a Sony QX1, it's about 640x480 and I am struggling to understand the datarate and bitrate settings. As I understand from the wiki, lower bitrate equals longer range. 
But how low can I go and how can I confirm that the txpower is actually maxed? (Atheros)

Does anyone know how much latency is added when displaying videofeed in QGroundcontrol via USB tether?

careyer, 2018-01-28:
Updated my above post... I now figured out where the video gets displayed... However the data received seems not to be in the anticipated format. @ivanskj are you familiar with this problem? How have you solved it with your Sony QX1?

ivanskj, 2018-01-29:
I have to check my notes but if i could remember correctly i did not use rtp pay/depay. I can post my complete gstreamer pipeline when I get home.

Are you using h264?

careyer, 2018-01-29:
@ivanskj : Thank you! **Your pipeline would be very much appreciated.** I think RTP is indeed the cause of trouble since it is a packet oriented network protocol usually supposed to go over UDP. Now by rethinking things Rodizios tx_rawsock seems to be a transparent tunnel for a bytestream (is that correct?). It just makes sense that the receiving side can't determine what the heck to do with a constant stream of bytes where no RTP packets can be determined easily.

Unfortunately I am not using h264 but a raw videostream.

I think I just have to leave out the rtp-elements in the gestreamer pipeline:
@TX: instead of : 
`gst-launch-1.0 v4l2src device=/dev/video3 ! "video/x-raw ..." ! rtpvrawpay ! filesink location=/dev/stdout`
@TX: use :
 `gst-launch-1.0 v4l2src device=/dev/video3 ! "video/x-raw ..." ! filesink location=/dev/stdout
`

and for the receiver side of things use:
@RX instead of: 
`gst-launch-1.0 filesrc location=/root/videofifo1 ! "application/x-rtp ..." ! rtpvrawdepay ! fbdevsink`
@RX use: 
`gst-launch-1.0 filesrc location=/root/videofifo1 ! "application/x-rtp ..." ! fbdevsink` or
`gst-launch-1.0 filesrc location=/root/videofifo1 ! "application/x-rtp ..." ! videoconvert ! fbdevsink`
Hoping that the fbdevsink element will deal with the input (autovideosink somehow can not be used without running an xserver).

Anyhow. Your pipeline is very much welcome. I am a absolute beginner with gstreamer and might just be missing the magic thingy.
Thanks

ivanskj, 2018-01-29:
Had a hard time figuring out how to open that raw .img file of the SD-card on my Mac, but i got it eventually!

My pipeline looks like this:
`gst-launch-1.0 souphttpsrc location=http://192.168.122.1:8080/liveview/liveviewstream is-live=true do-timestamp=true ! image/jpeg,width=640,height=424 ! jpegparse ! omxmjpegdec ! omxh264enc control-rate=1 target-bitrate=$BITRATE ! h264parse config-interval=5 ! fdsink | nice -n -9 /root/wifibroadcast/tx -p 0 -b $VIDEO_BLOCKS -r $VIDEO_FECS -f $VIDEO_BLOCKLENGTH -t $VIDEO_FRAMETYPE -d $VIDEO_WIFI_BITRATE -y 0 $NICS`

You would have to experiment with the input part of your pipeline since yours is a bit different than mine. I get a MJPEG stream from the ip-address of the camera. The jpegs are parsed, decoded and then encoded into a h264 stream. h264 is very good for streaming over a solution like this. Surely there could be many improvements done to this, but i am not sturdy enough on gstreamer to say, yet.

This is on v1.5 of WBC, so any adaptations to 1.6 needs to be considered. I am waiting for 1.6 final before I resume the hunt for long-range.

If you encounter any problems with the omx parts, let me know and i will check which version of gstreamer i am using because i remember experimenting with different gstreamer builds. Omx is the hardware accelerated framework used rpi.

Hope this helps in any way

careyer, 2018-01-29:
@ivanskj : Thanks! i wanted to avoid encoding the video to h264 but rather send the rawvideo... I will give it a try anyway. How does your decoding pipeline look like? Encoding seems to work but decoding gives me no output (but no error either)

ivanskj, 2018-01-29:
The RX is untouched since I am using H264. I wanted to test uncompressed MJPEG but i could not find a mission planner app that shows JPEG stream. H264 is good at recovering from packetloss.

What is the reason for using raw? 
What is displaying the recieved data?

ivanskj, 2018-01-29:
One way to test is to use filesink and try to play the file locally before you send it trough tx. It has to be package and muxed correctly, i.e mkv

ivanskj, 2018-01-29:
What type of format is /dev/video3 sending?

careyer, 2018-01-30:
@ivanskj: Oh man... i did not consider that hellovideo was a h264 player after all. I tried with gstreamer for hours and hours but it does not display anything if the datarate is too low (the thermal camera has a very negligible datarate due to its low resolution). Encoding to h264 and using hellovideo for playback it worked right out of the box :-) Thanks man!

Whatsoever I think raw video might be the better solution here because of two reasons:
1. the initial proof of concept using RTP/UDP had considerably less lag
2. the datarate of the h264 encoded 160x120 video is so tiny that WifiBroadcast even struggles at times 

@rodizio1 : Regarding No.2 the packet counter indicates that only very few packets do arrive. I think I may have to tweak the VIDEO_BLOCKS / VIDEO_FECS / VIDEO_BLOCKLENGTH variables to get less lag and more packets per timeframe arriving?
Here is how it looks right now:
[image] 
Video: https://youtu.be/5aDpvRuO9cI

###  Here is a little update on my attempt to transmit rawvideo instead:
I avoid using rtp in my gstreamer pipeline now and instead transfer only the raw datastream as is. 

I decided to keep things as simple as possible and instead of using my FLIR use the nifty little testvideo included with gstreamer. I first build a simple gstreamer pipeline (encode + decode & display) **localy on my GroundPi:**
`gst-launch-1.0 videotestsrc ! video/x-raw ! videoparse ! autovideoconvert ! fbdevsink`
This works flawlessly and yields the expected result: **a clean testpattern**: https://youtu.be/m8TwNFyYmSs

I then split up this working pipeline to my Air- and GroundPi as follows:
**AirPi:**  `gst-launch-1.0 videotestsrc ! video/x-raw ! fdsink fd=1 | tx_rawsock ...`
**GroundPi:**  
`gst-launch-1.0 filesrc location=/root/videofifo1 !  videoparse ! autovideoconvert ! fbdevsink`
This however results in **a distrubed Test-Pattern**: https://youtu.be/soXGVjsfk4A

I also did a short video to explain and demonstrate things. When transfering the video from my FLIR (this has a small osd showing °C) i can see parts of this osd flickering in the messed up output from time to time, i.e. data from the camera is arriving but for some reason gets mixed up.
[image] 
Video: https://youtu.be/4QUvAJTst7A

@rodizio1 : Any idea of what might be the reason for the rawvideo getting mixed up? Am I doing something wrong? I am so close to get this thing working - i do not want to give up on this. Thanks!

careyer, 2018-01-30:
@ivanskj 

> What  type of format is /dev/video3 sending?

I don't exactly know what format it is. I just know that this rtp pipeline worked like butter for it.

**TX**: `gst-launch-1.0 v4l2src device=/dev/video3 ! video/x-raw,width=160,height=128, framerate=1/10 ! rtpvrawpay ! udpsink host=192.168.0.100 port=9002`

**RX**: `gst-launch-1.0 udpsrc port=9002 caps="application/x-rtp,media=(string)video,clock-rate=(int)90000,encoding-name=(string)RAW, sampling=(string) YCbCr-4:2:0, depth=(string)8, width=(string)160, height=(string)128, colorimetry=(string)BT601-5, payload=(int) 96" ! rtpvrawdepay ! autovideosink`

ivanskj, 2018-01-30:
I had latency approx. 120-150ms when using h264encoding, you need to use the omx and check that you have sufficient gpu_mem if you go that route.

Maybe you are missing the pay/depay parts and the recieved data is not put together correctly? Try with the jpegpay element as a test.

rodizio1, 2018-01-30:
Sorry I have no idea about gstreamer.

Regarding the latency and wbc: with 8/4/1024 fec it'll wait until it has 8x1024bytes of data, then add fec packets and send them out. The rx waits until it has received those 8x1024 bytes packets (and the fec packets if datas were missing).

So if your flir only puts out let's say 3mbit instead of the default 6mbit, you'd either need to also halve the packet size or the number of packets to maintain the same wbc transmission latency. Or a combination of both. I'd try 6/3/750 or 4/2/750.

ivanskj, 2018-01-30:
Thank you for the clarification regarding FEC. That will definetly help alot!

I experienced RX overriding the FEC value to 1400 due to CTS_protection even when it was disabled in the config. Any pointers to why this could be happening?

careyer, 2018-01-30:
@rodizio1 : Thank you very much for the excellent explanation of `VIDEO_BLOCKS / VIDEO_FECS / VIDEO_BLOCKLENGTH` and how they act together. That makes perfect sense (and is much more intuitive to understand than the explanation in the Wiki ;-). You should consider adding this information there. It really helps a lot. 

Is there any way to test if some additional/unexpected data gets in between the rawvideo packets while being transmitted via tx_rawsock? It pretty much looks as if some additional bytes or something get injected here and therefore corrupt the rendering of the raw videoframes on the GroundPi. I believe it must be something like this because I can clearly see parts of the videotest pattern flickering at wrong positions in the output. As far as I understand tx_rawsock transfers a stream of data transparently to the rx, am I right? 

So in theory it should be possible to transform a pipeline like (simplified):
`(Pi1)rawvideo  -> (Pi1)display`  (i.e. locally displaying the video)
to:
`(Pi1)rawvideo -> (Pi1)tx_rawsock -> (Pi2)rx -> (Pi2)display` (i.e streaming video vom Pi1 to Pi2)

That is what i do with the above pipelines which results in a mixed up rendering of the content.

The reason why this works with h264 encoded frames is simple:
Even if additional/unexpected data is transferred together with the h264 frames via tx_rawsock the Ground Pi will only look for valid h264 frames in the output of rx, render them and discard the rest, whereas in a rawvideo scenario each additional byte gets interpreted as content of the video and such disturbes the anticipated output. Maybe something else (other than the rawvideo) from stdout gets into the rx_rawsock tunnel? 
I connected my AirPi to HDMI and observed that a packet counter of transmitted packets is displayed/incremented during transmission. Since this output is displayed on the console (stdout?) it gets transferred via tx_rawsock as well?

Using rawvideo on an anyhow low res video has the advantage that more data is transmitted (still a fraction of the data the HD-video stream consumes), lower latency because of lacking encoding, and after all EZ-Wifibroadcast seems to do a better job on higher datarates.

I can't tell how much I appreciate your help and support! ## Thanks Rodizio!!!

@ivanskj : Also many thanks to you guiding me to the h264 approach - at least things basically work now! BTW: I think I do not need the pay/depay part since i am not payloading to a network protocol here. I couldn't depay on the GroundPi (using hellovideo) anyway ;-)

ivanskj, 2018-01-30:
What i meant by using the pay/depay is that i believe you can force the groundPi to identify the correct packets and discarding the other info. Also by using them i believe the recieving Pi also knows where to start reading the image data. Without it all the recieved data may seem like unorganized bits and bytes.

careyer, 2018-01-30:
@ivanskj : Ahhh! I think I do understand now what you mean.... I was considering that you suggested to apply pay/depay to the already h264 encoded data. Now I understand that your idea is to shape the rawdata with pay/depay. Sorry I was missunderstanding you.

Indeed that is what I tried in the very beginning. Unfortunately gstreamer only allows payloading into RTP (Real Time Protocol) or GDP (Gstreamer Data Protocol) packets.
**TX**: `gst-launch-1.0 v4l2src device=/dev/video3 ! video/x-raw ! rtpvrawpay ! fdsink fd=1`

**RX**: `gst-launch-1.0 filesrc location=/root/videofifo ! video/x-raw ! rtpvrawdepay ! autovideosink`

However I believe `pay/depay `is not the reason why the current h264 solution works. My understanding is that pay/depay only loads and extracts the payload data into the above mentioned network protocol packets. However we do not need network protocol packets in a EZ-Wifibroadcast world ;-).

I think it is rather the `parse` part that makes it work because the parse part tries to identify a certain payload type (e.g. h264parse detects h264 encoded frames in a stream of data) and forwards it for processing, all the rest is discarded - so exactly what we want.

So in an ideal scenario the pipelines would look like this:
**TX**: `gst-launch-1.0 v4l2src device=/dev/video3 ! video/x-raw ! rtpvrawpay ! fdsink fd=1`

**RX**: `gst-launch-1.0 filesrc location=/root/videofifo ! video/x-raw ! rtpvrawdepay ! rawvideoparse ! autovideosink`

For whatever reason some of the Gstreamer elements which could do the trick such as:

- `rawvideoparse` (gstreamer-plugins-bad: Converts unformatted data streams into timestamped raw video frame)

- `unalignedvideoparse` (gstreamer-plugins-bad: Parse unaligned raw video data)

can not be found on my raspi even though I have installed gstreamer-plugins-good, gstreamer-plugins-bad, gstreamer-plugins-ugly completely. Any idea why those ones are missing?

Cheers!

3dpinta, 2018-01-30:
Which FlirOne version are you using (which gen and connector) and how are
you connecting it to the Pi? Also, which version of the Pi are you using
for Tx? I would love to do this as well as I have been looking at doing so
for a year with my Seek Thermal but that's not live video I can get
wirelessly.

ivanskj, 2018-01-30:
Now i understand some of my problems when i first started with this. Possibly lots of gibberish being sent along with the videodata.

Can you check if you have the correct sub-package of plugins-bad installed, using gst-inspect?
I compiled a newer version of gstreamer to get access to all the missing plugins. Thats a though job with alot of pits to fall into.

Would it be possible to pipe the raw output from the device without going through gstreamer?
i.e. `cat /dev/video3 | tx ....` ?

rodizio1, 2018-01-30:
@careyer What bitrate do you get from the flir cam? Is it correct, that it puts out 10 160x120 jpegs per second? That doesn't sound like much.

@ivanskj 
Maybe it makes sense to not use gstreamer at the TX at all, this way you'd be transferring just the jpegs over the wbc link, then all you'd need is some jpeg display program at the RX.

Regarding the pipes and stdout etc. Does gstreamer really output both it's console text output as well as the video on stdout? Have a hard time believing that, but who knows :)

ivanskj, 2018-01-30:
Thanks, I will test that out for sure. Maybe it will give a low latency solution when taking the encoding part out of the equation. 

I believe gstreamer has some environment variables controlling log-level. But yeah, sounds strange that it would output all that info to fdsink. It does print alot of other warning/debug info when something is configured wrongfully.

ivanskj, 2018-01-30:
Any tips to how i can measure the bitrate of the videofeed?

careyer, 2018-01-30:
@rodizio1 : Yes the FLIR One outputs 160x120 @ 10fps. If the output is in mjpeg format i cannot say for sure though. The resolution does not sound much but it is actually pretty decent for a consumer grade thermal camera. As you can image compressing that tiny amount of data with h264 even further results in a very very low datarate for EZ-Wifibroadcast to be transfered and causes some problems -e.g it takes longer as anticipated to fill all those datapackets before they are being transmitted. That is the main reason why I try to transmit raw (uncompressed) data at the end of the day, which results in a higher datarate. A proof of concept transmitting via RTP/UDP shows that this works indeed nicely with very little lag and no problems at all. Hoewever the aim is to go with the allmighty EZ-Wifibroadcast since it has many advantages ( you know the best) ;-).
My current implementation (using h264 compression for the sake of nessesity right now) with EZ-Wifiboradcast still suffers from more lag than the original RTP/UDP PoC - but has probably more range and robustness. However there is room to improve. Being able to fly with visible light and thermal vision on a multicopter has so much potential!

Recarding your question about gstreamer and piping stuff out on stdout: I think it really is true - there are several reports about that. That is why stdout is somewhat problematic. As @tbeauchant supposed it is a cleaner solution to use a custom fifo (generated with mkfifo) instead of stdout which sorts out all sorts of troubles. You never know what process or subprocess probably outputs random stuff to stdout at any given time. It would be a blaze to see something like this implemented (at least optionally via a parameter) in a future tx_rawsock. That would ensure an absolutely "clean" and transparent channel. 👍 👍 👍 

@ivanskj : How did you manage to compile the most recent versions of the gstreamer good, bad and ugly plugins? Can you give me some guidande how to do that? That would be very much appreciated. I would really like to give it a got with the more recent gstreamer elements like rawvideoparse and so on. :-)

@3dpinta : Using a Flir One Gen2 for Android (Micro USB), any Raspi will do but since connected via USB you should at least have two USB ports (WiFi + thermal camera). The Flir One Pro Gen3 is yet untested. It would be nice to know though if it works. The respective Linux driver has been reverse engeneered by some clever guys from eevblog. The driver in combination with the l4l2loopback kernel module provides two videostreams from the Flir: Thermal + HD visible light.

rodizio1, 2018-01-30:
> Recarding your question about gstreamer and piping stuff out on stdout: I think it really is true - there are several reports about that. That is why stdout is somewhat problematic. As @tbeauchant [...]

Sigh. Read-up again about pipes, stdin, stdout,stderr. tx_rawsock does _not_ mix the output data stream and it's status messages. That's actually not possible because it doesn't even send the datastream to stdout but to the wificard. The rx also doesn't suffer from this (there it would be possible because it actually does put out the datastream to stdout) because it simply prints status messages to stderr. I would've assumed gstreamer does the same that's why I didn't believe that such a stupid problem exists with it.

3dpinta, 2018-01-31:
@careyer

Thanks for the response. I may pick up a Gen3 and give it a try. Do you
have a good set of instructions on how to get it running with the Gen2? If
so, I can give it a go with the Gen3 once I pick one up.

rodizio1, 2018-01-31:
@careyer just looked at the video, if it's just around 80kbit, you need to reduce the packetsize and fec depth a lot more. I'd try something like 4/2/256 or 2/2/256, may need a little experimenting.

@ivanskj Regarding the bitrate measuring: 1.6 shows the live and max. measured bitrate on the osd. Or use 'pv'. The 1400byte packet size you see when cts is enabled was a 'hack' used in 1.5 to be able to use the same video bitrate (is was 6mbit fixed) for cts and non-cts modes. The CTS reduces bandwitdh, I basically 'bought it back' by using larger packets (which have less overhead and thus give more bandwidth).

In 1.6 this isn't the case anymore as the bandwidth gets measured, i.e. it will just be a little lower when using cts (and leaving the default 1024bytes packetsize).

ivanskj, 2018-01-31:
Sweet! I am now confident 1.6 will provide a robust solution. I’ll experiment with these new features.
Thanks for the info!

@careyer: I used various guides online to compile gstreamer and plugins. It was a painful process. I can look through my bookmarks and see if i can find it again. V1.6 uses newer kernels. Maybe it has updated repository for apt-get? Precompiled is alot easier if possible.

[63 more replies]